Atmosphere-NX/Atmosphere #372: CANT INSTALL NSP!!!!! on new fw 7.0

Nefith, 2019-02-22:
error when try to install nsps....

maciak, 2019-02-22:
You need new patches. Not available in AMS.

fincs, 2019-02-22:
No piracy supported here

closed 2019-02-22 by fincs

gjimenezf, 2019-02-24:
Even with ES and FS patches it is still not possible to install any nsp files: 
https://github.com/XorTroll/Goldleaf/issues/235#issuecomment-466674546

ZachyCatGames, 2019-02-24:
> Even with ES and FS patches it is still not possible to install any nsp files:
> [XorTroll/Goldleaf#235 (comment)](https://github.com/XorTroll/Goldleaf/issues/235#issuecomment-466674546)

Atmosphere doesn't support warez

gjimenezf, 2019-02-24:
It doesn't install Retroarch.nsp, Goldleaf.nsp they are not warez...

ZachyCatGames, 2019-02-24:
> It doesn't install Retroarch.nsp, Goldleaf.nsp they are not warez...

I read your previous message incorrectly

fincs, 2019-02-24:
Homebrew is distributed in NRO format (launchable with nx-hbloader + nx-hbmenu), not "NSP". Locking this issue.