Atmosphere-NX/Atmosphere #375: Atmosphere 0.8.4 won't boot

Kalos90, 2019-02-23:
Bug Report
What's the issue you encountered?
A couple of weeks ago I​ accidentally upgraded to firmware 7.0.0 via choidujour; now that Atmosphere 0.8.4 is out, I put it on my SD root, but it still won't boot. The Atmosphere logo is displayed, then nothing happens and the screen goes black. I have already tried using a fat 32 SD card with no results

How can the issue be reproduced?
Select fusee primary payload on bootloader

System Firmware Version
7.0.0

Environment?
What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee primary
Official release or unofficial build: Offical
Offical release version 0.8.4
Do you have additional kips or sysmodules you're loading: No
Homebrew software installed: None
Additional context?
Additional info about your environment: None

SciresM, 2019-02-23:
If you aren't seeing the sept logo, it means fusee-secondary isn't detecting that you have 7.0.0 installed. Did you use ChoiDujourNX? Are you sure that your package1 contains right contents?

Kalos90, 2019-02-23:
I see the Sept logo for a brief moment, then I see the atmosphere logo and then the screen goes black

SciresM, 2019-02-23:
Does error occur if you delete atmosphere folder and use a fresh installation?

Kinda sounds to me like bad firmware install.

Kalos90, 2019-02-23:
What do you mean by "fresh installation"? Like with previous atmosphere updates, I have pasted and overwritten the old folder on my SD root with the new release

SciresM, 2019-02-23:
I mean "delete atmosphere folder on SD" "drag atmosphere release zip to SD".

Btw you didn't answer: Did you use ChoiDujourNX?

Kalos90, 2019-02-23:
Yes, I have used choidujournx. I don't know what package1 is

Kalos90, 2019-02-23:
I tried a fresh installation, now it won't even load the Sept load. And the OS Loader often crashes

lildrill, 2019-02-24:
same issue here, only showing sept when i boot

ohsshoot, 2019-02-24:
Also having this issue.

acffordyce973, 2019-02-24:
Having the same issue. Updated via choidujournx, deleted atmoshphere folder and copied the latest. Sept shows, atmosphere logo, switch logo, then black screen. Boots fine on stock.

lildrill, 2019-02-24:
I got mine working again with reinstalling everything on the sd card

SciresM, 2019-02-24:
Can everyone in this issue verify that they don't have the issue that happens when one installs non-exfat firm but has an exfat sd card?

That's my best guess.

RRSlider, 2019-02-24:
I had the same issue. I didn't just delete and reinstall the atmosphere folder but the sept, modules, and bootloader folder as well. 
I used Kosmos' last release and they mentioned in the notes that if you don't do this, you might encounter a black screen. Seems to work for me now just fine. Not sure if this will be helpful.

Kalos90, 2019-02-24:
Where can I find kosmos latest release? I wanna give that a shot.

SciresM, how can I verify what you asked?

SciresM, 2019-02-24:
@Kalos90 if your SD card is formatted exFAT, try formatting it FAT32

tumGER, 2019-02-24:
>Where can I find kosmos latest release? I wanna give that a shot.

https://github.com/AtlasNX/Kosmos/releases/tag/v11.10.1

However what @RRSlider said only "fixes" the fact that your SD is messy and has artifacts everywhere. I seriously doubt this helps.

RRSlider, 2019-02-24:
I kind of doubted it would do anything but for the sake of covering all options.
I've only ever used the Kosmos releases y'see.

Kalos90, 2019-02-24:
@sciresM already tried, didn't work

SciresM, 2019-02-24:
@Kalos90 pretty sure this is due to SD card artifacts/mismatched FIRM vs sd card format...You really tried reformatting as FAT32 + using completely clean ams release zip fully extracted to sd root with nothing else?

Kalos90, 2019-02-24:
@SciresM yep, doesn't work

I have tried everything you guys suggested, but it still won't boot

SciresM, 2019-02-24:
Use Hekate to dump BOOT0/package2 partition? Guess you might have corrupted fw install somehow...

Kalos90, 2019-02-25:
I solved the issue by booting up with hekate..still having troubles getting some games to work, but that will improve with time I guess. Thanks y'all

closed 2019-02-25 by Kalos90