Atmosphere-NX/Atmosphere #377: Feature request: Block Nintendo servers

gjimenezf, 2019-02-24:
Please integrate an option to block Nintendo servers to avoid ban as SX OS does.
90 DNS does not works for me.

leo60228, 2019-02-24:
@DavidBuchanan314 is working on this, but I don't think it will be included in Atmosphère.

designgears, 2019-03-07:
There is already a great solution for this, would be nice to see something integrated tho.
https://gbatemp.net/threads/incognito.531924/

designgears, 2019-03-07:
Why was that deleted?

fennectech, 2019-03-07:
People dont trust blarwar

fincs, 2019-03-07:
Please do not link to closed source tools produced by people of questionable reputation.

fennectech, 2019-03-07:
i would like to see it possable to disable specific domains as I’ve been disabling only the telemetry domains allowing me to ‘safely’ play online. (Have done it for over a year without a ban)

designgears, 2019-03-07:
> People dont trust blawar

People don't like him, I haven't seen him post anything malicious outside of a naming dispute and some other silly drama. But Ok.

fennectech, 2019-03-07:
It wasn’t a dispute.  He basically stole the name tinfoil from an existing project.

closed 2019-05-23 by SciresM