Atmosphere-NX/Atmosphere #379: Atmosphere Crash after starting N64 Emulation

Klauserus, 2019-02-26:
## Bug Report

AMS Crash

### What's the issue you encountered?

When i load content with Mupen64 Next Emulator (RetroArch) = Blackscreen and than an error screen from AMS.

### How can the issue be reproduced?

Use Retroarch 1.7.6 or the dayly Nightly, load core mupen64 Next and a n64 rom you want.

### Crash Report

Error Code 2168-0002 (0x4a8)
Title: 010000000000100D
Firmware: 7.0.1 (Atmosphere 0.8.4-master-2673118)
PC: 000000001A4548B4 Backtrace: 000000001A2D4000

### System Firmware Version

Switch 7.0.1
AMS 8.4
Retroarch 1.7.6 26.02.18 Nightly

### Environment?

- What bootloader: Hekate 4.8

Thank you

aks about more information, if you need

SciresM, 2019-02-26:
Your crash report is incomplete.

SciresM, 2019-02-26:
Also, it's probable that n64 emulation/retroarch need access to main application memory, and thus you might need to override an application instead of an applet.

Klauserus, 2019-02-26:
You need the crash picture? I make one....
you mean about the applet, i need a native nsp version of retroarch?

fincs, 2019-02-26:
No. Override an eShop demo with the native HBL support in AMS. Don't install nsps. There is no reason whatsoever for their existence and they can get your console banned.

drofdoom, 2019-02-26:
You have to use the nsp forwarder to have enough ram to run it launched
from almbum it does not have enough resources .

ZachyCatGames, 2019-02-26:
> You have to use the nsp forwarder to have enough ram to run it launched from almbum it does not have enough resources .
> […](#)

This is the recommended method 
https://gbatemp.net/threads/use-atmosphere-to-access-full-ram-with-homebrews-without-nsp.521240/

NSPs are bad

fincs, 2019-02-26:
>You have to use the nsp forwarder to have enough ram to run it launched

No you don't. Please actually read my comment - you can override an eShop demo or any other official APPLICATION title.

hippydave, 2019-02-26:
^ Definitely do what they're saying. Also you need to use m4xw's special build of mupen next (check his twitter or whatever), the ones in nightlies aren't current until libnx is updated.

ZachyCatGames, 2019-02-26:
> ^ Definitely do what they're saying. Also you need to use m4xw's special build of mupen next (check his twitter or whatever), the ones in nightlies aren't current until libnx is updated.

you dont need m4xw’s special build. But it does certainly offer a smoother experience:P

hippydave, 2019-02-26:
> > ^ Definitely do what they're saying. Also you need to use m4xw's special build of mupen next (check his twitter or whatever), the ones in nightlies aren't current until libnx is updated.
> 
> you dont need m4xw’s special build. But it does certainly offer a smoother experience:P

I mean yeah if your end goal is playable N64 games, you need it.

closed 2019-02-26 by SciresM