Atmosphere-NX/Atmosphere #380: First reboot from power menu after replacing reboot_payload.bin boots to Atmosphere

Deuxgod, 2019-02-27:
## Bug Report

### What's the issue you encountered?

Using the latest binary, after replacing the stock reboot_payload.bin with another payload using FTP and booting from power menu, loads atmosphere instead of the selected payload. Doing a new reboot from power menu will reboot to the new payload.

### How can the issue be reproduced?

Replace reboot_payload.bin from Atmosphere with any other payload (I used Argon NX payload) using FTP or USB and try to reboot from power menu, it will boot back to Atmosphere. Doing a second reboot will boot to the newly designated payload.

### System Firmware Version

7.0.1</br>

### Environment?

Booted with primary payload using the latest official release from Github.

aveao, 2019-02-27:
I've checked the code and it seems like it's loaded on init of bpc_mitm's reboot manager: https://github.com/Atmosphere-NX/Atmosphere/blob/67c7ef69f4d986ea132e4a7434cd95bc3e5e3a8e/stratosphere/ams_mitm/source/bpc_mitm/bpcmitm_reboot_manager.cpp#L32

I feel like just using reboot-to-payload homebrew in cases like this would make more sense, as it does load the payload from SD before it reboots (and doesn't rely on the one bpc_mitm loaded on init).

Deuxgod, 2019-02-27:
> I've checked the code and it seems like it's loaded on init of bpc_mitm's reboot manager:
> 
> [Atmosphere/stratosphere/ams_mitm/source/bpc_mitm/bpcmitm_reboot_manager.cpp](https://github.com/Atmosphere-NX/Atmosphere/blob/67c7ef69f4d986ea132e4a7434cd95bc3e5e3a8e/stratosphere/ams_mitm/source/bpc_mitm/bpcmitm_reboot_manager.cpp#L32)
> 
> Line 32 in [67c7ef6](/Atmosphere-NX/Atmosphere/commit/67c7ef69f4d986ea132e4a7434cd95bc3e5e3a8e)
> 
>  if (R_FAILED(Utils::OpenSdFile("/atmosphere/reboot_payload.bin", FS_OPEN_READ, &payload_file))) { 
> I feel like just using reboot-to-payload homebrew included with ams in cases like this would make more sense, as it does load the payload from SD before it reboots (and doesn't rely on the one bpc_mitm loaded on init).

Yeah I noticed after that it's loaded at boot and not on runtime, so it's not really a bug, I'm sorry

SciresM, 2019-02-27:
Yeah, closing because not-a-bug.

closed 2019-02-27 by SciresM