Atmosphere-NX/Atmosphere #382: Atmosphere 0.84 won't boot

Elbelejuibe, 2019-03-02:
Atmosphere 0.84 won't boot on my switch 7.0.0.

Hekate launch correctly : I can print Fuses info or TSEC keys normally but when i launch Atmosphere, I can see the sept logo, the atmosphere logo, the nintendo logo and a black screen until the switch logo.

I use a 64gb SD card format in exFAT.

tslater2006, 2019-03-02:
I had this issue, problem was I didnt have the exfat firmware installed. Atmosphere cant boot if Horizon can't read the SD card. Heres how I solved...Format it to fat32, boot atmosphere, install exfat version of firmware with Choi. Then reformatted to exfat

ZachyCatGames, 2019-03-02:
> I had this issue, problem was I didnt have the exfat firmware installed. Atmosphere cant boot if Horizon can't read the SD card. Heres how I solved...Format it to fat32, boot atmosphere, install exfat version of firmware with Choi. Then reformatted to exfat

Or format to fat32 and keep it formatted as fat32...

tslater2006, 2019-03-02:
I was assuming they had it exfat for a reason and wanted to keep it exfat. But yep that works too :)

bottmint, 2019-03-03:
There's no reason for having a card intended for the switch formatted as exfat, outside of restoring a nand backup.

fincs, 2019-03-03:
HOS' exFAT firmware/driver is *extremely* unreliable & has a tendency to corrupt microSD cards randomly. For this reason it is heavily discouraged to install exFAT firmware or use microSD cards bigger than 32GB because they'll be formatted with exFAT.

bottmint, 2019-03-03:
3rd party utilities do exist that can format cards larger than 32gb as fat32 on windows, and linux formatting utilities can do it out of the box. i don't know about mac os.

TheCoder3156, 2019-03-03:
@Elbelejuibe please use the issue template.

closed 2019-03-03 by SciresM