Atmosphere-NX/Atmosphere #384: Atmosphere crashes when overclocking CPU

Crusatyr, 2019-03-05:
## Bug Report

### What's the issue you encountered?

Testing done by AliciaBurrito was reported at this site: https://gbatemp.net/threads/freebird-switch-clock-speed-control-sysmodule.530641/page-7#post-8510124

I do believe that provides all the relevant details on the issue

### How can the issue be reproduced?

By using Freebird,or sys-clk, increase the CPU overclock to max for near immediate results. Some reports claim Atmosphere is unstable at any CPU speeds higher than 1.2Ghz, however

### Crash Report

Don't have one

### System Firmware Version

6.2.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - Official Release 0.8.3 and 0.8.4
- Do you have additional kips or sysmodules you're loading: n/a
- Homebrew software installed: N/A

### Additional context?

TheCoder3156, 2019-03-05:
I think this is because atmosphere runs more custom sysmodules than something like ReiNX and therefore requires more stability. Overclocking any cpu will cause stability issues at some level especially when doing more cpu intensive things (like more sysmodules in this case). I’m not sure if everything I said is 100% true and these are just my thoughts.

SciresM, 2019-03-05:
It's probably just that AMS runs more on the syscore --> race conditions hit more often at higher clicks.

HerptyDerpoty, 2019-03-07:
You guys just going to double down on race conditions that don't exist or lies that make atmosphere sound like the second coming of CHRIST? Race conditions that exist at one speed exist at all speeds since everything in running on the same cpu, not to mention that the switch itself changes clock speeds and I don't see anything in your code or hekate that specifically addresses that so.
You zealots should just fix your software.

SciresM, 2019-03-07:
Ignoring the rest of the content in that comment...it is my understanding that changing the speed at which a processor operates will change the relative timings of events, which can cause whether race conditions occur to change.

It's an honest best guess, not like there's any other plausible reason I can see.

HerptyDerpoty, 2019-03-10:
How you can even make such a beyond wrong claim when you claim to be a programmer blows my mind, that is not how race conditions work at all, 2 processes executing on the same processor will have the same complete times relative to the processors clock, meaning if a race exists between 2 processes on the same cpu that race will exist at all clock rates and the only thing effected is how often it happen, and the fact that it happens not at all at stock speeds but within minutes at non stock speeds on atmo means that the likelyhood that a 100mhz oc (10% of stock) causing such a dramatic increase in the rate at which crashing happens is approaching 0%.

ALSO the claim that the reason it works on rei because of bad hwinit (something m4wx claims and sticks too) is bs because rei uses old hekate hwinit when it was just a carbon copy of nintendos hwinit.

the plausible answer is atmo has a bug reinx doesn't have

SciresM, 2019-03-14:
@Crusatyr "EDIT: It was determined in RS discord that Exo is not at fault here and fatal does not generate a report since it is not called"

This is false, pretty sure it's an exosphere problem (it's not a coincidence that reinx began experiencing this issue on 7.0.0, where it uses exosphere).

Anyway pretty sure I found the problem, testing it atm.

commit 3fcad4bc (2019-03-14): exo: fix SE driver coherency bug (closes #384)
closed 2019-03-14 by SciresM