Atmosphere-NX/Atmosphere #387: CAL0 read protection doesn’t seem to be working

ThatNerdyPikachu, 2019-03-08:
## Bug Report

### What's the issue you encountered?

A friend decided to try an app that shall not be named that reads from CAL0 using standard FS calls. This user did not have the flag that allows a title to read it defined, but the app was still able to read it.

### How can the issue be reproduced?

1. Don’t set the CAL0 read flag on the HBL title
1. Try and read from PRODINFO

### Crash Report

N/A

### System Firmware Version

7.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: 0.8.5
- Do you have additional kips or sysmodules you're loading: N/A

### Additional context?

N/A

SciresM, 2019-03-08:
@ThatNerdyPikachu What title was used for HBL? This is actually relevant to the logic.

SciresM, 2019-03-08:
@ThatNerdyPikachu pushed a possible fix....there was a big issue with AMS's flag detection. Check again?

ThatNerdyPikachu, 2019-03-10:
seems to be working, will reopen if anything else arrises.

closed 2019-03-10 by ThatNerdyPikachu