Atmosphere-NX/Atmosphere #390: Atmosphere Auto-Updater

Essasetic, 2019-03-08:
### What feature are you suggesting?
#### Overview:
- [Have an option in the System Settings > System > Under System Update to have a box that says "Update Atmosphere CFW" (or something along the lines of that).]</br>

#### Smaller Details: 
- [It could be as simple as coding in the releases page of the Atmosphere GitHub into the box so it can grab the latest releases when they come out.]</br>

### What component do you feel this would best fit within?
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's custom bootloader.</br>

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
If this is a feature for an end-user, how does it benefit the end-user? 
It would benefit the end user by removing the inconvenience of turning off the switch, getting the update off GitHub, putting it onto the SD Card, injecting the payload and finally booting back in.</br>

TheCoder3156, 2019-03-08:
While this is a great idea I doubt this should be for fusée. Other than that this should be a thing.

Essasetic, 2019-03-08:
> 
> 
> While this is a great idea I doubt this should be for fusée. Other than that this should be a thing.

What do you think it should be for then? I'll change it to the appropriate module.

TheCoder3156, 2019-03-08:
Probably troposphere or it's own separate homebrew.

Essasetic, 2019-03-08:
> 
> 
> Probably troposphere or it's own separate homebrew.

Fixed.

sandhoners123, 2019-06-04:
It should be part of loader or a kernel patch maybe (?)

sandhoners123, 2019-06-04:
Like amsu (atmosphere updater) service

Mikau28, 2019-06-20:
Would it be possible to do this when using hekate? Then after updating reboots back into the payload?

fennectech, 2019-06-21:
If it doesn’t stomp on reboot-payload.bin it shouldn’t cause an issue other
than updating fusee-primary.bin Wich can be sidestepped with an ini
pointing to /atmosphere/fusee-primary.bin

ELY3M, 2019-06-24:
I think it would be best to do updater as separate homebrew.   you need to update payload bin too.   
some people have their payload.bin stored somewhere on their sd card or trinket m0 mod chip.   
need to set up a path to payload bin if you store the payload bin on your sd card.   
everyone have different setups.

fennectech, 2019-06-24:
we should update a fusee-primary.bin stored on sd card as well for those who use Hekate to chainload.

j-james, 2019-10-29:
This doesn't look like it's received much traction. I'll just throw out that @ITotalJustice created an auto-updater [as an NRO](https://github.com/ITotalJustice/atmosphere-updater) recently. Whether it should remain as a separate application or become part of Troposphere is up to the more knowledgeable :grin:

Essasetic, 2019-10-29:
> This doesn't look like it's received much traction. I'll just throw out that @ITotalJustice created an auto-updater [as an NRO](https://github.com/ITotalJustice/atmosphere-updater) recently. Whether [...]

I'm satisfied with this. However, I'll leave this open just incase anybody wants to make an updater for atmosphere without any external homebrew apps.

demios97, 2020-10-29:
A year has passed since the last message and the new version of atmosphere ceased to support the update through similar homebrews. Devs, please, give us another solution.

fennectech, 2020-10-30:
Deepsea_updater and hekate_updater work just fine.

bakugo, 2020-10-30:
@demios97 the solution is to update manually. Not everything should be done for you.

tarikalmashni, 2021-01-03:
There's the "All in One Updater" homebrew (find it on gbatemp or github) which does the job pretty well.
Since it's also able to download sigpatches, I won't link to it here though.

ELY3M, 2021-01-17:
I am working with someone who only have phone. He do not have pc at all.     there need to be a way for person to update atmosphere via cfw.

SciresM, 2021-01-17:
You can't extract a zip file to a console's SD card using only a phone, so you can't reasonably have installed atmosphere in the first place if you're phone only.

That's not a use case I'm interested in catering support to.

If you're interested in hacking your $300 game console, and you really have no other computer whatsoever, consider picking up a raspberry pi for like $10-30.

closed 2021-01-17 by SciresM

myzar472, 2021-01-17:
Hey, uh... @SciresM? I think you closed the wrong issue. That last comment was unrelated to the original Issue above requesting an updater for Atmosphere.

EmreTech, 2021-01-17:
This is unnecessary. There is a homebrew app called AIO Switch Updater that can update Atmosphere already.

SciresM, 2021-01-17:
@myzar472 the original request ("add a button inside qlaunch") is technically infeasible, and atmosphere doesn't support updating while HOS is alive anyway, so I closed this issue, yes.

myzar472, 2021-01-17:
> This is unnecessary. There is a homebrew app called AIO Switch Updater that can update Atmosphere already.

It almost seems to me that having this be handled by third parties can open things up to clashes with SciresM's vision for Atmosphere. You aren't wrong about other solutions existing, but you're advocating a homebrew application that also downloads copyrighted firmware and sigpatches right now.

EmreTech, 2021-01-17:
That is true, but I thought I could mention it. Anyways, nothing beats updating Atmosphere manually.

tarikalmashni, 2021-01-17:
@SciresM (or whoever deleted my comment): I'd like to know why my comment has been deleted since I wrote the very same thing @EmreTech literally wrote 4 comments later: An app like AiO Updater exists which serves this exact desire.

@ELY3M Using your phone with your Switch is pretty easy. You have two options:

1. Use an OTG adapter with an SD card reader (I guess there are even adapters that take the card directly) so they appear as a USB device on your phone (depends on your phone though).
2. This method works with **every phone** and doesn't require any additional hardware: You create a WiFi hotspot from your phone and connect your Switch to it. Then you use FTPD on your Switch. On the phone side you use something like ES File Explorer or any other FTP client to connect to your Switch FTP server created this way and push the files over.
Obviously it's nicer if Switch and phone are connected to the same home WiFi instead of having to create a hotspot for that.

fincs, 2021-01-17:
As pointed to by @SciresM, hot updating Atmosphère while Horizon OS is running is not a supported use case.

References to the suggested "updater" application have additionally been removed, because it contains a warez (sigpatch) feature, as @myzar472 said.

Locking down this issue.