Atmosphere-NX/Atmosphere #392: Cheats can never be enabled

locomain, 2019-03-09:
### What's the issue you encountered?

When having this setting dmnt_cheats_enabled_by_default = u8!0x1 the cheats do not enable.
If i open Edizon  and press y a popup shows up with the following message: Atmosphere cheat module is not running on this System cheat managment and variable freezing is disabled.

### How can the issue be reproduced?

By setting dmnt_cheats_enabled_by_default to u8!0x1

### System Firmware Version

7.0.1

### Environment?

Official release using hekate boatloader

Essasetic, 2019-03-15:
> 
> ### What's the issue you encountered?
> 
> When having this setting dmnt_cheats_enabled_by_default = u8!0x1 the cheats do not enable.
> If i open Edizon and press y a popup shows up with the following message: Atmosphere cheat module is not running on this System cheat managment and variable freezing is disabled.
> ### How can the issue be reproduced?
> 
> By setting dmnt_cheats_enabled_by_default to u8!0x1
> ### System Firmware Version
> 
> 7.0.1
> ### Environment?
> 
> Official release 0.8.5 using hekate boatloader

Did you overwrite the Atmosphere files to update?

SciresM, 2019-03-15:
Cannot reproduce. Sounds like you didn't have dmnt module on SD, though.

closed 2019-03-15 by SciresM