Atmosphere-NX/Atmosphere #393: Sciresm is a dick

LilPuff420, 2019-03-09:
You're also gay. why else would you banned me from the server on discord so quick after I said that three letter word? Kill yourself you useless piece of shit.

fennectech, 2019-03-09:
Lol

LilPuff420, 2019-03-09:
Not when his bitch ass mind take shit out of context

LilPuff420, 2019-03-09:
SciresM is gay. I'm kinda glad I got banned. 
I don't need to be dealing with a MALE sucking off another MALE.

zkitX, 2019-03-09:
Rule # 2
> 2. Be nice to each other. It's fine to disagree, it's not fine to insult or attack other people.
>  • You may disagree with anyone or anything you like, but you should try to keep it to opinions, and not > people. Avoid vitriol.
> • Constant antagonistic behavior is considered uncivil and appropriate action will be taken.
> • The use of derogatory slurs -- sexist, racist, homophobic, transphobic, or otherwise -- is unacceptable > and may be grounds for an immediate ban.

>homophobic slurs

We do not want you here. @SciresM please close this.

closed 2019-03-09 by TuxSH