Atmosphere-NX/Atmosphere #458: overwrite multiple titles with hbl

justinjustin7, 2019-03-10:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Allow multiple titles to be used to launch hbl.

#### Smaller Details: 
- Use loader.ini to specify multiple titles to be used for launching hbl.

#### Nature of Request:  
- Change

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
This will allow for use of the new cheat engine through overwriting a system title while still benefiting from the increased memory of overwriting a game title. May also open the possibility of using the cheat engine with homebrew?

bottmint, 2019-03-10:
As an immediate solution, make a copy of your hbl.nsp, put it in `atmosphere/titles/010000000000100D` and rename it to `exefs.nsp` to have hbmenu on the album applet, and use loader.ini to overwrite a game title.

justinjustin7, 2019-03-10:
Nice, that works. It still would be good to have a more straightforward way of doing it through the ini though.

Crissal1995, 2019-03-10:
In atmosphere/loader.ini change title_id to "app", and every game will instead boot hbl with the combination you use

bottmint, 2019-03-10:
I believe what @justinjustin7 is wanting is to be able to define a game title to overwrite _in addition to_ being able to define and applet to overwrite in the loader.ini, unless I'm mistaken.

SciresM, 2019-03-22:
The use case that wants this should be totally resolved by 9e1e9ff8.

closed 2019-03-22 by SciresM