Atmosphere-NX/Atmosphere #459: 1.0.0 compatibility issues

yuukieve, 2019-03-10:
## Bug Report
### What's the issue you encountered?
Atmosphère has two compatibility issues with firmware version 1.0.0:
 - The `fatal` reimplementation has never worked on 1.0.0, and any crashes simply lead to a blank black screen; the button triggers do not work to reboot, and you must hard power off.
 - Past commit 4711d05, Atmosphère does not boot on firmware 1.0.0, instead getting stuck at the "Nintendo Switch" splash screen.

### How can the issue be reproduced?

- `fatal` breakage: Boot Atmosphère commit 4711d05 or earlier and cause a crash; observe `fatal` does not start.
- Hang at boot: Boot Atmosphère commit 4711d05 or later, observe that it gets stuck at the "Nintendo Switch" logo. 

### Crash Report

These are not issues that cause crashes, but even if they were, 1.0.0 does not have `creport`, and thus no crash report can be provided.

### System Firmware Version

1.0.0 

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: 
  - `fatal` breakage: All commits/releases since the initial reimpl; `fatal` has never been compatible with 1.0.0. 
  - All commits/releases past/including commit 4711d05 lead to a hang at boot.

### Additional context?

- This has previously been discussed on Discord, and this issue was posted with SciresM's approval.

fincs, 2019-03-10:
For reference, `fatal` compatibility with 2.x was recently solved (in e62606d2762441d0b8da25b4ce6ecafd1150cc1d).

yuukieve, 2019-03-10:
After cherrypicking e62606d, I can confirm that it fixes `fatal` compatibility with 1.0.0 as well. However, the issue with commit 4711d05 remains.

commit 20ba6432 (2019-03-25): set.mitm: fix 1.0.0 compat (closes #459)
closed 2019-03-25 by SciresM