Atmosphere-NX/Atmosphere #460: Flags dont seem to be working for cal0 access

lerptato, 2019-03-11:
Trying to use certain homebrew and cal_read and bis_write flags don't seem to allow cal0 access, either does hbl_bis_write and hbl_cal_read for that matter

aveao, 2019-03-11:
Please use the issue template.

bakugo, 2019-03-11:
You're probably having the same issue as https://github.com/Atmosphere-NX/Atmosphere/issues/387, which has already been fixed in https://github.com/Atmosphere-NX/Atmosphere/commit/66560d0a7b06203edf1879476da604dfcc1f85bc

The protection logic is inverted in the current release, so if you want your homebrew to have those flags, just remove the flag files (at least until the next release).

lerptato, 2019-03-11:
I've tried with and without the flags, it wont give me write access either way

SciresM, 2019-03-11:
@bakugo it's not quite inverted, heh...

If you want the flags to *not* be present, place both flags on the SD card.

If you want the flags to be present, place 0-1 of the two flags on the SD card.

(Where the two flags are /%s.flag, and /flags/%s.flag)

Or just wait for 0.8.6, which I'm sure will be soon.

ThatNerdyPikachu, 2019-03-11:
FWIW, there is no CAL0 write flag.

aveao, 2019-03-11:
Yep, relevant commit and comment:

https://github.com/Atmosphere-NX/Atmosphere/commit/a00e120bf7be1a1407db89d6e184ad35fcf53adf

There's no valid reason for a homebrew app to be able to write to CAL0, IMO. Too risky, regardless of usecase.

ThatNerdyPikachu, 2019-03-11:
If you ***really*** want to write CAL0, just remove the Partition::PRODINFO case, and make the last else always return a ProxyStorage — not a ROProxyStorage.

lerptato, 2019-03-12:
how do it go about that? i'm essentially trying to 0 out my prod info and downgrading is too much of a hastle

closed 2019-03-12 by lerptato