Atmosphere-NX/Atmosphere #480: Games install on atmosphere 0.8.0 won't run on atmosphere 0.8.4

Lassall, 2019-03-16:
Error message: corrupted data
Examples: Super Smash Bros Ultimate and Mario Party

fincs, 2019-03-16:
> Games install

Are these eShop downloads? If so, then try popping in a fresh new SD card & redownloading the games from your eShop list of purchases.

Lassall, 2019-03-16:
> 
> 
> > Games install
> 
> Are these eShop downloads? If so, then try popping in a fresh new SD card & redownloading the games from your eShop list of purchases.

I think they all are. But only the two mentioned games won't work. The other games are running fine.

fincs, 2019-03-16:
Still, I think it would be good if you tried a clean install of those games fresh from the eShop, to rule out SD card corruption.

Lassall, 2019-03-16:
At first, thank you for your support! I'll try that. What constellation/combination would you recommend?
I have firmware version 6.2.0. Will that work stable with AMS 0.8.5 and which TinFoil/Goldleaf version should I use?

profi200, 2019-03-16:
Atmosphere or it's devs don't support piracy.

Lassall, 2019-03-16:
> 
> 
> Atmosphere or it's devs don't support piracy.

Who said anything about piracy, I didn't ask for games or download links?! I'm new to the scene and really don't know what tools I need to run my game backups from SD-card. Sadly theres not only one knowledgebase where everything is properly described. I'll make one myself if I know enough.There are plenty of tools and versions.  The scene is like "Here you go with our new whatever version, teach yourself what you can do with it".

Daniel0Wellby, 2019-03-16:
Atmosphere's devs don't support piracy, at all

closed 2019-03-16 by fincs