Atmosphere-NX/Atmosphere #485: Unable to wake from sleep 7.0.1

pixeltester, 2019-03-22:
## Bug Report

### What's the issue you encountered?

After updating from firmware 7.0.1 from 6.2.0 using ChoiDujourNX I am unable to wake from sleep. The home button will wake the screen which remains blank. Power or Volume will reboot to payload. If I roll back to 6.2.0 sleep mode is fully functional again.

### Crash Report

No crash reports generated.

### System Firmware Version

7.0.1

### Environment?

Official Atmosphere release 0.8.5, launched with fusee-primary, no additional kips or sysmodules.

### Additional context?

I have had active parental controls since 4.1.0 (to retain the pin unlock) however I have kept my Switch offline since then, could it be related to parental controls?

SciresM, 2019-03-22:
Huh. I have been sleeping/waking on 7.0.1 fine using latest release.

WinterMute, 2019-03-22:
sleep/wake is also working fine for me on 7.0.1

pixeltester, 2019-03-22:
It's very strange, could it be related to parental controls? I can remove them but (I think) I'll have to do a full NAND restore of my 4.1.0 backup to get them back which is why I haven't tested that yet.

fincs, 2019-03-22:
Same. Sleep/wake working flawlessly on 7.0.1 and latest stable ams.

SleepyMan, 2019-03-23:
Same problem here. It might be related to an error on ChoiDujourNX about failing to write SAFE pkg1, etc., at the end, but continuing to reboot either way. ChoiDujourNX probably can't write anything to BOOT0 due to AMS protection, and thus gives these errors.

Just to clarify, I updated to 7.0.1 from 6.2.0, using ChoiDujourNX 1.0.1 and Atmosphere 0.8.5-7ddaad6, from Mar 19, and it gave the above errors at the end. On, 7.0.1, fails to launch any title (original games, so not related to sigpatches, etc), but homebrew was OK. Sleep problem like the one mentioned here. Downgraded to 6.2.0, and gave the same errors on writing SAFE pkg1, etc. Everything working normally on 6.2.0.

fincs, 2019-03-23:
I ask you to test the same thing: restore last NAND backup done before using Choi and use the official system update function to get to 7.0.1.

ThatNerdyPikachu, 2019-03-23:
That error was probably fixed with https://github.com/Atmosphere-NX/Atmosphere/commit/9d5ca47ac8a2afe2fc1f2f67d460efc96bb15902#diff-0ac19e73964583e3b3784132f5d73b5bR169

pixeltester, 2019-03-23:
> 
> 
> Same problem here. It might be related to an error on ChoiDujourNX about failing to write SAFE pkg1, etc., at the end, but continuing to reboot either way. ChoiDujourNX probably can't write anything to BOOT0 due to AMS protection, and thus gives these errors.

I have the same errors when upgrading 6.2.0 > 7.0.1 but it does show some writes to Boot0 before the errors, no errors at all when downgrading back to 6.2.0. I also have the possibly unrelated title launching error you mentioned.

So is the cause of the sleep issue a problem with a ChoiDujourNX upgrade to 7.0.1?

These are the errors in a ChoiDujourNX 6.2.0 to 7.0.1 update for reference:

Error 0x313802 writing SAFE pkg1 to Boot1 offset 0x0 size 0x40000!
Error 0x313802 writing SAFE pkg1 to Boot1 offset 0x40000 size 0x40000!
Error 0x313802 writing offset 0x0 of NORMAL pkg2 to BCPKG2-1-Normal-Main!
Error 0x313802 writing offset 0x0 of NORMAL pkg2 to BCPKG2-2-Normal-Sub!
Error 0x313802 writing offset 0x0 of SAFE pkg2 to BCPKG2-4-SafeMode-Main!
Error 0x313802 writing offset 0x0 of SAFE pkg2 to BCPKG2-4-SafeMode-Sub!

> 
> 
> I ask you to test the same thing: restore last NAND backup done before using Choi and use the official system update function to get to 7.0.1.

This is a reasonable suggestion but I would rather avoid a full NAND restore if possible and as a homebrew user I prefer to keep my Switch offline and perform offline firmware updates, so it will be impractical in the future (and possibly ban worthy if Nintendo logs multiple updates from 4.1.0?) to keep flashing a clean NAND for every official firmware update to get around this problem.

fincs, 2019-03-23:
> This is a reasonable suggestion but I would rather avoid a full NAND restore if possible and as a homebrew user I prefer to keep my Switch offline and perform offline firmware updates, so it will be [...]

Unless you have corrupted your Switch by installing unsigned titles (which is something that has **no** place in homebrew), there is nothing to worry about going online. Atmosphère blocks error upload and also disables a good portion of telemetry. I don't think multiple upgrades from 4.1.0 can be a bannable offense, and it *could* happen for legitimate users if the user interrupts the update procedure and ends up with corrupted firmware, forcing them to use safe mode to resume/restart the upgrade process.

WinterMute, 2019-03-23:
> This is a reasonable suggestion but I would rather avoid a full NAND restore if possible and as a homebrew user I prefer to keep my Switch offline and perform offline firmware updates, so it will be [...]

You could (and probably should) make another NAND backup after updating, restoring to the same backup to perform all future updates is kind of silly. I'm not sure I see the point of staying offline in case you get banned - how is that different from being banned?

pixeltester, 2019-03-23:
> You could (and probably should) make another NAND backup after updating, restoring to the same backup to perform all future updates is kind of silly. I'm not sure I see the point of staying offline [...]

There is no need for a new NAND backup after updating, my original NAND backup is fine for safety purposes. Whether it's from 4.1.0 or any other firmware the main point I was making is that doing that restoring a NAND that has never used homebrew as a workaround is impractical.

Many homebrew users keep their units offline. In my case my main Nintendo account is on the unit for the parental control pin unlock and I would prefer to err on the side of caution.

It seems SciresM has accounted for the ChoiDujourNX incompatibility in his commit yesterday so I'll wait to try that in the next official release, I don't need 7.0.1 immediately.

UnicornsOnLSD, 2019-03-23:
Is your fusee primary up to date? The sleep mode library necessary for sleep mode may be missing (this only affects autoRCM users iirc)

fincs, 2019-03-23:
> The sleep mode library necessary for sleep mode may be missing 

What? I don't recall fusée needing a separate sleep mode library...

jakibaki, 2019-03-23:
Fusee definitely doesn't need a sleep library, that's just a hekate thing.

pixeltester, 2019-03-23:
I decided to go ahead and test a build with the latest commits, the result was that ChoiDujourNX now only shows the following two errors for the SafeMode Firmware package1 instead of the six errors I listed earlier:

Error 0x313802 writing SAFE pkg1 to Boot1 offset 0x0 size 0x40000!
Error 0x313802 writing SAFE pkg1 to Boot1 offset 0x40000 size 0x40000!

After updating to 7.0.1 wake from sleep works again and there is no title launching error either.

SciresM, 2019-03-23:
@pixeltester ugh, do I need to give it boot1 access, to? Guess I'll do that.

commit 047f3b65 (2019-03-23): fs.mitm: also give choinx access to boot1 (closes #485)
closed 2019-03-23 by SciresM