Atmosphere-NX/Atmosphere #486: Exfat warning on boot

fennectech, 2019-03-22:
## Feature Request

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ If you do not have the information needed for a section, replace its contents with "Unknown". ]</br>
[ Lines between [ ] (square brackets) are to be removed before posting. ]

[ Please search for existing [feature requests](https://github.com/Atmosphere-NX/Atmosphere/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3A%22features%2Ffeature-request%22) before you make your own request. ]</br>
[ Duplicate requests will be marked as such and you will be referred to the original request. ]

### What feature are you suggesting?
#### Overview:
- [ Prompt a the user on boot if they are using exfat telling them it isnt recommended listing the issues it will cause ]</br>

#### Smaller Details: 
- [ Trigger the error message on boot if possable.  ]</br>

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Addition

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
[ it would benefit the user because it would tell them WHY their file system keeps corrupting. ]</br>
[ It would keep developers from answering issues and questions about problems with filesystem corruption ]</br>

fennectech, 2019-03-22:
My suggestion is for an error message such as “We have detected you are using the ExFat filesystm on your uSD card.   This has been known to cause problems on the switch.  Please do not ask for support for ExFat related issues. A tool like fat32format can be used to format your card to fat32 and should solve your sdcard corrupting issues.

bakugo, 2019-03-22:
Please no. 4 mandatory splash screens on boot are already more than enough.

And when was the last time someone opened an issue here that was caused by exfat?

fennectech, 2019-03-22:
We closed one earlier today.   I’m talking about a message that appears if and only if the users is using exfat.   If they are using fat32 they dont get it.

ThatNerdyPikachu, 2019-03-22:
But what if the user hasn’t experienced any issues? What would be the point?

fennectech, 2019-03-22:
Because.   They will.   Its just a mater of time.

SciresM, 2019-03-23:
let's not.

closed 2019-03-23 by SciresM