Atmosphere-NX/Atmosphere #487: Multiple title_id replacements

CecilFF4, 2019-03-22:
## Feature Request

#### Overview:
- A lot of people don't want to install NSPs for their apps, for one reason or another, which is why it's nice to have a title replacement option through atmosphere. Typically the album is replaced. However, it would be nice to have both that AND an option to replace a title app, so certain apps (like Retroarch) can use full RAM mode while we retain the ability to access HBL through the album. Another issue I've encountered is that some people have problems with Edizon when running through title replacement, so they need to keep that album replacement functionality.

#### Nature of Request:  
- Change
  - Allow the loader.ini [hbl_config] to have more than one title_id replacement.

### Why would this feature be useful?
- See overview section.

thedax, 2019-03-22:
You can already do this in 0.8.5 stable with a workaround by making a copy of hbl.nsp, renaming it to exefs.nsp, and putting it in /atmosphere/titles/010000000000100D, and using title_id=app in your loader ini.

Also, this was addressed more properly earlier today in commit https://github.com/Atmosphere-NX/Atmosphere/commit/9e1e9ff8c0e68c48c375abfe0c8cfe8a89b80ae8, which will make it so the default title id is the album, but it'll also override any apps when you hold R.

SciresM, 2019-03-22:
Duplicate of #458, which I closed earlier today with 9e1e9ff8.

closed 2019-03-22 by SciresM

fincs, 2019-03-22:
> A lot of people don't want to install NSPs for their apps, for one reason or another, which is why it's nice to have a title replacement option through atmosphere.

I'd like to clarify that NSP is **not** the way people should be running their homebrew, and it isn't and will never be sanctioned by the development teams behind devkitA64 and libnx.

CecilFF4, 2019-03-22:
I've been messing with the "fix" noted in #458, but I can't get it to work. I've tried this:

> [hbl_config]
title_id=010000000000100D
path=atmosphere/hbl.nsp
override_key=R

> [hbl_config]
title_id=01007ef00011e000
path=atmosphere/exefs.nsp
override_key=R

I've tried this:

> [hbl_config]
title_id=010000000000100D
path=atmosphere/hbl.nsp
override_key=R
title_id=01007ef00011e000
path=atmosphere/exefs.nsp
override_key=R

Neither work. Yes, my hbl.nsp was copied and renamed. I'm just looking to run HBL by holding R for either album or BOTW, depending on my needs.

CecilFF4, 2019-03-22:
This didn't work either:

> [hbl_config]
title_id="app"
path=atmosphere/hbl.nsp
override_key=R

thedax, 2019-03-22:
You can't have two entries like that. Set it to BOTW's title id, like this:

[hbl_config]
title_id=01007ef00011e000
path=atmosphere/hbl.nsp
override_key=R

(or `title_id=app` with no quotes if you want it to be for every app/game)

and then copy the exefs.nsp (renamed hbl.nsp) to `/atmosphere/titles/010000000000100D` like I said in my prior reply. This will let you launch it from either place, but the hotkey to control the album will be L instead, because it's controlled by layeredFS.

If having a different hotkey isn't acceptable, then you'll need to self-build Atmosphere or wait until the next release, since setting both hotkeys to the same button doesn't work, to my knowledge.

SciresM, 2019-03-22:
Just use the config I set to be the new default...I have confirmed both games and album get overridden when holding R.

CecilFF4, 2019-03-22:
This didn't work:

> [hbl_config]
title_id=app
path=atmosphere/hbl.nsp
override_key=R

I'm perfectly fine with every app being overridden, so this is a fine solution, except it didn't work. I couldn't access HBL from any app, including album. Sorry that this is not working for me, despite your efforts. I'll work on thedax's solution using layeredfs next.

SciresM, 2019-03-22:
Solution I mentioned involves a commit added today. Just wait for next atmosphere release if you want stuff to be simple, or build it yourself.