Atmosphere-NX/Atmosphere #489: Feature: Language emulation though set.mitm

randompasser, 2019-03-25:
### What feature are you suggesting?
#### Overview:
Allow the user to override the language and region settings used by a game using `config.ini`

#### Smaller Details: 
`set` provides commands for applications and applets to obtain the currently set language and region of the console. This feature would allow an end user to override values returned by `set` on a per app basis via set.mitm by specifying the desired language in the title's `config.ini`

For example, the follow would override the language to Japanese with the console region set to Japan:

```
[set_override]
override_language=ja
override_region=jpn
```
[Switchbrew page](https://switchbrew.org/wiki/Settings_services#set)
[libnx GetLanguageCode implementation](https://github.com/switchbrew/libnx/blob/11b7bece4aa96b8e9abdc46f3d3e20ccbe6a4585/nx/source/services/set.c#L120)
[libnx GetLanguageCode sample](https://github.com/switchbrew/switch-examples/tree/master/settings/get_system_language)

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
Selecting a new language in qlaunch requires a full system reboot, I know reboot to payload is a thing but jumping between languages can get tedious especially with the increased boot times on 7.0.0+ due to sept. 

This would be a huge quality of life improvement for people who speak multiple languages or trying to pick up a new one if it does get implemented.

SciresM, 2019-04-13:
Yeah, I can do this one.

XargonWan, 2019-05-22:
Is it possible to let every user choose its language?

SciresM, 2019-05-22:
Done.

commit 6f25e928 (2019-05-22): set.mitm: language emulation (closes #489)
closed 2019-05-22 by SciresM