Atmosphere-NX/Atmosphere #491: Game crash causes console to get stuck on a black screen

bakugo, 2019-03-28:
## Bug Report

### What's the issue you encountered?

I've been playing around with a glitch in BOTW (latest version) that can consistently crash the game. If done on stock with no CFW, the normal "software was closed because an error occurred" screen pops up and I can go back to the home menu. If done while running atmosphere, the console instead freezes on a black screen, and must be manually shut down by holding the power button.

Removing the creport (0100000000000036) folder from the SD card stops this from happening.

### How can the issue be reproduced?

Not sure if this happens with all game crashes or just this one. I don't know how to crash the game in any other way, or how to crash other games.

To reproduce the exact crash, follow this guide: https://www.youtube.com/watch?v=cL_DAl63zxg
Except when you gain control of link while also controlling the gyro, jump into the pit. The game should get stuck with the camera looking at the pit. Press start and select "return to title screen" and the game should crash after the loading screen.

### Crash Report

No crash report is generated.

### System Firmware Version

7.0.1

### Environment?

Fresh installation of atmosphere 0.8.6, booted with fusee-primary. Also happened in 0.8.5.

### Additional context?

None.

fennectech, 2019-03-29:
ive seen this for a while and its infuriating.

aveao, 2019-03-29:
I get freezes on Trials Legends crashes but get a proper game crash (software was closed because an error occurred) on Sonic Mania.

Running on commit ca42a9d, OFW 7.0.1.

hippydave, 2019-03-29:
I get freezes on Celeste too (reproducable if v patient, Celeste seems to always crash after playing it for hours). Will test without creport next time I have a long Celeste session.

Peacock1090, 2019-04-02:
Was playing Chocobo's Mystery Dungeon, then was going to charge the Switch at the dock and come back to it later. When I came back it had a crash for some reason. Rebooted it to payload and it now hangs after the Nintendo Switch logo at bootup.

Doubt messing with creport will fix it so waiting til someone who knows what they're actually doing tells me how tp fix it basically.

hippydave, 2019-04-02:
> Was playing Chocobo's Mystery Dungeon, then was going to charge the Switch at the dock and come back to it later. When I came back it had a crash for some reason. Rebooted it to payload and it now hangs after the Nintendo Switch logo at bootup.

Not the same issue if you had the error screen with the reboot to payload option.

Peacock1090, 2019-04-02:
> Not the same issue if you had the error screen with the reboot to payload option.

never mind, this is a different issue and I need to use the actual template to discuss it here. I've been having trouble reading things due to dumb irl issues no one has the time to go thru obviously so disregard everything I said.

commit d0659377 (2019-04-13): creport: speed up code region detection (closes #491)
closed 2019-04-13 by SciresM