Atmosphere-NX/Atmosphere #492: sort the home menu items

toxic9, 2019-03-29:
will it be possible to implement a way to sort the home menu items?
It's so boring not to have an official way to do that.
No alphabetical order, no custom order... just a stupid order that keeps changing when you click an item.
I hope you will bring this in the future for the homebrew comunity!
an many thanks for you work.

hippydave, 2019-03-29:
Patches to qlaunch (home menu) might be in the scope of Troposphère, but you've also got people working on custom qlaunch replacements, so that might come sooner.

misson20000, 2019-03-29:
I think you can reorder them with ns commands? Would probably be pretty easy to keep them from being reordered after that.

SciresM, 2019-03-29:
The order is "most recently accessed, in any way".

Anyway this sounds more like an organizational homebrew thing than an ams thing, for now.

closed 2019-03-29 by SciresM