Atmosphere-NX/Atmosphere #494: Feature Request: ad-hoc emulation

Skalke, 2019-04-01:
## Ad-Hoc emulation

### What feature are you suggesting?
#### Overview:
- Emulation of Ad-Hoc in game, this could make some games without official online playable (for example Civilization VI).</br>

#### Smaller Details: 
- Currect method is a replace fs_mitm by ldn_mitm, but now it is going with some bugs cause of dmnt:cheat and fs_mitm refactor. I think the better way to rebuild ldn_mitm.</br>

#### Nature of Request:  
- Addition [ldn_mitm by Space](https://github.com/spacemeowx2/ldn_mitm) license [GPL-2.0](https://github.com/spacemeowx2/ldn_mitm/blob/master/LICENSE)
- Change fs.mitm?
- Removal dmnt:cheat? (may create critical freeze in Edizon with currect latest ldn_mitm).

### What component do you feel this would best fit within?

- [Exosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Fully-featured custom secure monitor.</br>
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- Better use it in Stratosphere if in future u will have HOS-sysmodules manager (like KipSelect)/

### Why would this feature be useful?
Nintendo Switch Online is not so cool as I imaged this, with custom servers we can create better territory-based lobbys with local community. It cause much less lags and troubles in a lot of games. </br>
</br>

SciresM, 2019-04-01:
Just add ldn mitm as a non kip sysmodule? I don't understand this request.

Skalke, 2019-04-01:
Now ldn_mitm is bugged, and i don't there is Space. it requered some updates and refactor for normal working with latest atmospheres

SciresM, 2019-04-01:
Sounds like something you should talk to the developer about...?

Skalke, 2019-04-01:
yeah. but i can't find developer online and don't know why dmnt:cheat is bugging ldn_mitm.

SciresM, 2019-04-01:
Sounds like an issue with ldn_mitm, which we do not maintain.

SciresM, 2019-04-04:
Closing for now (since no plans to merge ldn_mitm, and certainly not to do so without being able to talk to the dev due to maintenance concerns).

closed 2019-04-04 by SciresM

leo60228, 2019-05-13:
The ldn_mitm is confusing, however Civ 6 is one of the few games with LAN support and no online support. I think he was requesting some sort of Hamachi-like P2P VPN, which is likely out of scope anyway.