Atmosphere-NX/Atmosphere #496: HB Menu Freeze

drrastovac, 2019-04-04:
I have a problem after holding R and go to album sometimes I am getting random freeze screen and I must hold power 12 sec to restart switch. Is there any fix for that? I am using fresh Atmosphere 0.86 and I am on the latest 7.01 software.
Everything except that working flawlessly.

WinterMute, 2019-04-04:
Make sure you continue holding R until HB menu boots. Atmosphere has a bunch of modules that check the input at several points before the app finally boots and if your trigger button is pressed for one check but released for another then things get mismatched & sometimes cause crashes/hangs.

Sometimes if you leave it for a bit you'll get the Atmosphere crash report screen & you can reboot from there.

SciresM, 2019-04-04:
Closing because it sounds like user-error, if it turns out this is a real ams error lemme know and re-open.

Also, this doesn't use the issue templates, so I'd close it anyway.

closed 2019-04-04 by SciresM

drrastovac, 2019-04-04:
The user Master Kek also reports same behavior. It seems that I am not alone.

ZachyCatGames, 2019-04-04:
> The user Master Kek also reports the same behavior. It seems that I am not alone.

yea, I do occasionally have freezes like the ones you describe, they dont happen very often (like 2 times per month at most), I think they might just be crashes, as my most recent one generated a crash report

drrastovac, 2019-04-04:
I had a few freezes in last couple days. Formated my sd card and tried everything and getting the same error. Next time i will wait for crash report and i will screnshoot it.

fennectech, 2019-04-05:
I see this every once in a while.   This was not the user error stated above. I was not holding the button at all.

drrastovac, 2019-04-07:
> I see this every once in a while. This was not the user error stated above. I was not holding the button at all. Additionally. I let it sit fifteen minutes and it never went to the atmo reboot screen.

Same, I waited about 10 minutes for the crash log and nothing happened. Restarted my switch by holding power button couple of sec.