Atmosphere-NX/Atmosphere #504: Crash during boot phase

generiskk, 2019-04-09:
## Bug Report

Crash that happens during boot.

### What's the issue you encountered?

No changes done, vanilla installation from releases.

### How can the issue be reproduced?

Just starting up the system, let it run to the Nintendo logo, then the Switch logo, then stuck from there. 
### Crash Report

https://gist.github.com/generiskk/b60225abce97f024ea5a120a28330aa9

### System Firmware Version

7.0.1

### Environment?

Hetake & Atmosphère 0.8.6.
Or just sending Fusee_Atmosphere.bin

### Additional context?
N/A

SciresM, 2019-04-09:
Is your SD formatted correctly as FAT32?

generiskk, 2019-04-09:
Yes, tested with 3 different sd cards formated as fat32.

SciresM, 2019-04-09:
Does the error occur booting fully stock fusee (non-hekate?)

generiskk, 2019-04-09:
Took one of my sd cards, wiped it clean and did a "long" format of Fat32 instead of quick.

Fresh copy of atmosphere from here, and using fusse-primary.bin payload via TegraRcmGUI 2.5.

Freezes at the Nintendo switch logo. 

[01514810299_0100000000000016.log](https://github.com/Atmosphere-NX/Atmosphere/files/3058942/01514810299_0100000000000016.log)

[01514810299_0100000000000016_thread_info.bin](https://file.io/ZCSQEx)

fincs, 2019-04-09:
It's crashing in the wlan system module, huh?

generiskk, 2019-04-09:
If that is the issue, any way for me to disable the wlan system module? It is not needed as the system will not go online anyways.

fincs, 2019-04-09:
Have you done any hardware modification to your console/have you opened it up?

generiskk, 2019-04-09:
No hardware modifications done to the system, only changed plastic backcover so I can see witch switch system it is, have two of 'em.

fincs, 2019-04-09:
It's possible you might have done some damage to your console by doing that; possibly loosened the wifi antenna (not sure if it works like that) until it disconnected altogether.

generiskk, 2019-04-09:
I have nand backups, they have been tested, currently using the first nand backup I made before even booting into the Switch with CFW installed.

It's still spewing out errors. Is it at all possible to even disable the module?

fincs, 2019-04-09:
I have no idea; and that pretty much confirms hardware damage :(

camidumas, 2019-04-15:
I have had a similar error on 6.2 trying to use Atmosphere 0.8.6.  It froze on the logo.  I rolled back to 0.8.5 and everything was normal.  I don't have logs, sorry, I'm just mentioning this as it was a similar problem that definitely was not associated with hardware problems.

XargonWan, 2019-05-05:
I have the same issue, every first boot it's a crash, then I restart by volume button and it's working.

profi200, 2019-05-06:
Loose WiFi antenna won't do anything (if it doesn't short anything). It's probably a general hardware problem.

fennectech, 2019-05-11:
If its just the antenna come loose not the WiFi module (If there is a module) that will affect nothing. It will just get the shittiest reception you ever did see.    You can use fsmitm to disable the sysmodule by loading a dummy sysmodule under its title ID but how the rest of HOS will react to wlan sysmodule vanishing is quite unclear.  It would most likely crash.

generiskk, 2019-05-11:
> If its just the antenna come loose not the WiFi module (If there is a module) that will affect nothing. It will just get the shittiest reception you ever did see. You can use fsmitm to disable the sysmodule by loading a dummy sysmodule under its title ID but how the rest of HOS will react to wlan sysmodule vanishing is quite unclear. It would most likely crash.
> 
> You would do this by creating a dummy sysmodule NSP and loading it with the title ID 0100000000000016 in the same way you would load sys-ftpd or something. This would effectively replace the wlan sysmodule with the one from your sdcard wich is a dummy. Can you get into vanilla HOS? No hekate. No fusee no atmo at all (and burn fuses if you’ve saved em )

I will sadly have to put that on hold as I have damaged the sd-card reader module, I will have to come back on this on a later date.

fennectech, 2019-05-11:
Yowch.  Good luck fixing it.  Ether way.  It sounds like your console has
been through hell.  Might wanna sell it for 100$ on ebay and put it twards
a new tablet bit :)

generiskk, 2019-05-11:
> Yowch. Good luck fixing it. Ether way. It sounds like your console has been through hell. Might wanna sell it for 100$ on ebay and put it twards a new tablet bit :) Ether way. Dont just toss the tablet. Parts or not working ones sell for a good price. . In the mean time. If somebody were to create a dummy exefs NSP with that titleID I’d be glad to report on what HOS does if wlan is missing
> -- FennecTECH

Not a big issue, its the connectors on the sd-card adapter itself, not the mainboard. Will keep you updated when its resolved. Have ordered new parts ;)

closed 2019-05-23 by SciresM