Atmosphere-NX/Atmosphere #506: Crash after pressing home button in softwares like tinfoil

TachibanaSuzume, 2019-04-12:
## Bug Report

### What's the issue you encountered?

Crash after pressing home button in softwares like tinfoil

### How can the issue be reproduced?

Boot into Atmosphère 0.8.7, plug in the usb cable and connect to PC,  open tinfoil or lithium, press home.

### Crash Report

https://gist.github.com/TachibanaSuzume/70f27d0b56019d3348b11053ea70b856

### System Firmware Version

7.0.1

### Environment?

Hetake & Atmosphère 0.8.7.
Or just sending Fusee_Atmosphere.bin

### Additional context?

N/A

fincs, 2019-04-12:
Piracy tools like tinfoil or lithium are not supported.

closed 2019-04-12 by fincs

neonsea, 2019-04-12:
Wouldn't it still make sense to investigate the underlying cause? How are you guaranteeing this symptom is unique to "piracy tools"? Seeing as it's related to USB, I don't see why this can't affect other, legitimate homebrew not related to piracy.

fincs, 2019-04-12:
This is not an AMS issue, and very likely poor/sloppy coding on the piracy tool's behalf.