Atmosphere-NX/Atmosphere #509: Homebrew USB Issues

Adubbz, 2019-04-13:
## Bug Report

### What's the issue you encountered?

A crash occurs upon exiting any homebrew that attempts to use USB, with a USB cable connected. It is important to emphasize that this is not homebrew-specific. I have included my own test homebrew for this, however it has also been reported to impact:

* nx-shell (https://github.com/joel16/NX-Shell/issues/49)
* Goldleaf (https://github.com/Atmosphere-NX/Atmosphere/issues/508)
* Tinfoil/Lithium (https://github.com/Atmosphere-NX/Atmosphere/issues/506)

When overriding the album applet, the crash occurs immediately upon pressing the home button. When overriding a game, the crash occurs upon closing the game. When overriding the album applet, only a fatal report is produced. When overriding a game, both a fatal report and a crash report is produced.

Other users reported this issue as not occurring on ReiNX, however from my own testing, ReiNX simply appears to freeze rather than show fatal.

This issue may either be an Atmosphere issue, or a libnx issue. It's difficult to determine definitively, however I think it's important that it is documented on one of the two issue trackers.

### How can the issue be reproduced?

1. Open the homebrew with a USB cable connected.
2. Press the home button
3. (If launched using an application) Close the application.

Note that returning to the homebrew menu by pressing B (or whatever the homebrew has set it up to be), and *then* pressing home will sometimes prevent crashes, but this isn't reliable, and often simply delays it to happening later (for example, when launching a new game).

### Crash Report

Crash Report:
[01555115384_0100000000000006.log](https://github.com/Atmosphere-NX/Atmosphere/files/3075550/01555115384_0100000000000006.log)

Fatal Report:
[01555115380_0100000000000006.log](https://github.com/Atmosphere-NX/Atmosphere/files/3075551/01555115380_0100000000000006.log)

### System Firmware Version

7.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: 
  - 0.8.7

Otherwise, Atmosphere is a clean install.

### Additional context?

My test homebrew:
[DebugLogger..zip](https://github.com/Atmosphere-NX/Atmosphere/files/3075519/DebugLogger.zip)

fincs, 2019-04-13:
I'm able to confirm this using a USB test program of my own I had written some time ago.
My money is on this being a libnx bug; the usb-comms code hasn't been touched in a long time and it's possible that either 6.x/7.x slightly broke libnx usb, or we've never handled this situation properly. Does this actually work (i.e. not give a "Success" fatal error on USB) in previous versions of Atmosphère?

```c
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <switch.h>

void fail(void)
{
	while (appletMainLoop())
	{
		hidScanInput();
		u64 kDown = hidKeysDown(CONTROLLER_P1_AUTO);
		if (kDown & KEY_PLUS)
			break;

		consoleUpdate(NULL);
	}
}

int main(int argc, char* argv[])
{
	consoleInit(NULL);

	printf("Hello World!\n");

	Result rc = usbCommsInitialize();
	if (R_FAILED(rc))
	{
		printf("usbCommsInitialize: %08X\n", rc);
		fail();
	}
	else
	{
		printf("Now listening!\n");
		consoleUpdate(NULL);

		size_t retsize;
		u32 num;
		while ((retsize = usbCommsRead(&num, sizeof(num))) == sizeof(num))
		{
			printf("Received number: %u\n", num);
			consoleUpdate(NULL);
			if (num == 0)
				break;
		}

		usbCommsExit();

		printf("retsize ended up being %zu\n", retsize);
		consoleUpdate(NULL);
		fail();
	}

	consoleExit(NULL);
	return 0;
}
```

Also please do not link or mention warez tools.

Chimecho, 2019-04-13:
Related https://github.com/Atmosphere-NX/Atmosphere/issues/506

Adubbz, 2019-04-13:
This definitely has worked in the past, as it worked when I was originally playing with USB stuff on 6.0.0. As for exactly when it stopped working I'm unsure. I'm told that supposedly it still works on 6.2.0, so it may indeed be a 7.0.0-specific thing, but I have yet to get confirmation of this or check myself sadly.

fincs, 2019-04-13:
In that case more research/debugging of the USB sysmodule is required.

Also, to everyone posting references to warez tools or issues that were closed because they referenced warez tools: please stop. Warez tools are **not** supported by Atmosphère.

bakugo, 2019-04-13:
Dude chill out for a second. You are on a constant crusade against the piracy boogeyman and it's reaching a point where you're actively discouraging people from reporting bugs. If you don't want piracy tools mentioned just edit the post, don't rush to say "not an AMS issue" and lock because you might be wrong (like you were here).

We're all here because we want to see the software improve, there's no need to be hostile.

XorTroll, 2019-04-13:
People reference some Warez apps just because some of them actually use USB, same way they would reference non-warez apps, like NX-Shell in this case.

Anyway, I've been getting this issue with USB for a while with Goldleaf, and expected it to be a libnx issue, maybe comming from new 7.x changes in USB sysmodule.

fincs, 2019-04-13:
People should understand that warez apps are not supported nor catered to and they will never be, so any mention of them will not solicit any kind of attention from us - in other words, it's a waste of time. If someone like OP can prove that there is an issue that affects normal legitimate homebrew then we can actually start investigating.

Chimecho, 2019-04-13:
The referenced issue was not a "warez tool problem", but a USB one, even if the OP didn't know. That's why I referenced it.

ness151, 2019-04-14:
> People should understand that warez apps are not supported nor catered to and they will never be, so any mention of them will not solicit any kind of attention from us - in other words, it's a waste of time to talk about them. If someone like OP can prove that there is an issue that affects normal legitimate homebrew then we can actually start investigating.

I get this issue with NX-Shell, when trying to set up non-warez apps without ejecting my microsd.

designgears, 2019-04-21:
This appears to be resolved on firmware 8.0.0, or I'm getting extremely lucky.

IAmTheRealLinkle, 2019-04-23:
@designgears, do you atmo 0.8.8 for firmware 8.0.0 or in general 8.0.0 firmware?

designgears, 2019-04-23:
> @designgears, do you mean atmosphere 0.8.8 for firmware 8.0.0 or in general 8.0.0 firmware?

Both, atmosphere 0.8.8 is the only thing that can boot 8.0.0 and run homebrew. :)

james-d-elliott, 2019-04-24:
Using Adubbz' DebugLogger (libnx:3dde214) on AMS0.8.8/HOS6.2.0:

1. USB Plugged in, Python Script Running (python3)
2. Open Album (override with hbl)
3. Open DebugLogger and wait for the test output
4. Press home button
5. Switch exits normally to home screen.

SciresM, 2019-04-24:
So, uh, is this resolved(?)

SciresM, 2019-04-24:
I guess we probably want to solve this problem on lower firmwares if possible...

james-d-elliott, 2019-04-24:
Probably good to test on 7.0.1 again as well; and I can test any other HOS versions if needed. I attached the original source from Adubbz with the version compiled with libnx:3dde214 for anyone else who wants to test.

[DebugLogger.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3110583/DebugLogger.zip)

newget, 2019-04-24:
Not resolved on 7.0.1 with atmosphere 0.8.8
Tested with both NX shell and adubbz's test homebrew.

IAmTheRealLinkle, 2019-04-24:
Not sure if it matters, I did update from 5.1.0, to 8.0.0, with Atmo 0.8.8. USB is working perfectly now. Not sure what happened, but something was definitely up before.

fincs, 2019-04-24:
¯\_(ツ)_/¯ 

I'm going to go ahead and suppose this is a Nintendo bug; otherwise it doesn't make any sense.

james-d-elliott, 2019-04-24:
Adubbz had the same conclusion, I'll probably do some controlled testing later to verify.

averne, 2019-04-24:
Same here: https://github.com/joel16/NX-Shell/issues/48.

james-d-elliott, 2019-04-24:
So looking at the issues reported for this on this repo and various other USB-using repos it looks like a 7.0.1 issue. The only issues reported on 8.0.0 is slowness.. also looks like a libnx commit may have fixed it on 7.0.1, which Joel16 used (thanks @averne) - I'm waiting to hear from Joel which commit he used. 

I've also compiled the DebugLogger with the latest commit from libnx at this time (bc1786b) and [attached it](https://github.com/Atmosphere-NX/Atmosphere/files/3114451/DebugLogger.zip). So if @newget can check again that'd be great if you're still on HOS7.0.1. I'll try with 3.0.1 and 5.1.0 later today.

[DebugLogger-libnx_bc1786b.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3114451/DebugLogger.zip)

fincs, 2019-04-24:
I think we are now in a position where this issue can be closed.

closed 2019-04-24 by fincs

newget, 2019-04-25:
@james-d-elliott same result with DebugLogger-libnx_bc1786b
On 7.0.1 with atmosphere 0.8.8, I launched and quit with home button while usb cable plugged in.
Fatal error 2000-0000