Atmosphere-NX/Atmosphere #510: Per-title HBL override config

friedkeenan, 2019-04-14:
## Feature Request
### What feature are you suggesting?
#### Overview:
- Possible duplicate of #329 
- The ability to have different hbloader override keys for different titles</br>

#### Smaller Details: 
- Perhaps an `hbl` section in the config.ini under /atmosphere/titles/TITLE_ID/ that lets you specify an override key for hbloader in addition to the override key for LayeredFS</br>

#### Nature of Request:  
- Addition
  - An extension of the per-title LayeredFS override key system, but for hbloader (as talked about in **Smaller Details**)

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
I like holding R to override a game with hbloader, but I very much prefer not holding R to override the album with hbloader, and I expect many other end users feel the same. Having to hold R to override the album with hbloader isn't that much of a hassle, but it's *just* enough hassle to be annoying.</br>

james-d-elliott, 2019-04-14:
Maybe instead make them separate overrides. For example:

override_key_any_app=R
override_key_title_id=!R

Alternatively remove override_any_app and replace it with override_key_any_app. So a full hbl_config would look like this (I actually think I like this more, easier to implement and probably going to least affect users):

title_id=010000000000100D
path=atmosphere/hbl.nsp
override_key=!R
override_key_any_app=R

james-d-elliott, 2019-04-15:
Hopefully this commit doesn't break anything; my knowledge of the code base is very limited; biggest uncertainty is with Utils::IsHblTid, IsHBLTitleId, and ams_mitm. I attached a build of it (built w/ https://github.com/switchbrew/libnx/commit/3dde2148cc1635eae13fff7f7205099a7ffd01e3). But it works in testing using the config quoted below:

https://github.com/james-d-elliott/Atmosphere/commit/321169308a82c5742536832c020f419406234d2c

[fusee-secondary.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3078193/fusee-secondary.zip)

`[hbl_config]
title_id=010000000000100D
path=atmosphere/hbl.nsp
override_key=!R
override_key_any_app=R

[default_config]
override_key=!L
cheat_enable_key=L`

james-d-elliott, 2019-04-15:
Hopefully this commit doesn't break anything; my knowledge of the code base is very limited; biggest uncertainty is with Utils::IsHblTid, IsHBLTitleId, and ams_mitm. I attached a build of it (built w/ https://github.com/switchbrew/libnx/commit/3dde2148cc1635eae13fff7f7205099a7ffd01e3) **WARNING:** Use at your own risk, consider it alpha or worse. But it works in testing using the config quoted below:

https://github.com/james-d-elliott/Atmosphere/commit/321169308a82c5742536832c020f419406234d2c [fusee-secondary.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3078193/fusee-secondary.zip)

```[hbl_config]
title_id=010000000000100D
path=atmosphere/hbl.nsp
override_key=!R
override_key_any_app=R

[default_config]
override_key=!L
cheat_enable_key=L
```

znxDomain, 2019-05-14:
@james-d-elliott does your fork allow you to have a different `path=atmosphere/hbl.nsp` per title id?

Is that feature a side effect of this feature request? or do I need a new feature request for that?

james-d-elliott, 2019-05-23:
@znxDomain no it shouldn't but that should not be hard to do. It's unrelated to this issue though. This issue is more about the behavior or override_any_app I believe.

SciresM, 2019-05-23:
title-specific hbl configuration is a no -- hbl configuration should be global.

However, separating the key for overriding apps and overriding a specific title (album) could probably be done.

hippydave, 2019-05-24:
> However, separating the key for overriding apps and overriding a specific title (album) could probably be done.

This would be nice to have.

SciresM, 2019-12-31:
Implemented separated config for title/all applications (and for up to eight specific titles instead of one) a few weeks ago.

closed 2019-12-31 by SciresM