Atmosphere-NX/Atmosphere #511: Unable to boot recovery/maintenance mode on 7.0.1 & AMS 0.8.4 through 0.8.7

thedax, 2019-04-16:
## Bug Report

### What's the issue you encountered?

Trying to boot into maintenance/recovery mode results in the console hanging on the Switch (Joycon + nintendo switch) logo. Issue persists with both Hekate and Fusee-primary.

### How can the issue be reproduced?

Send fusee-primary or hekate and observe that maintenance mode will not boot.

### Crash Report

N/A

### System Firmware Version

7.0.1

### Environment?

Bootloaders; Hekate & Fusee-primary (as in, f-p launched on its own, not chainloaded)
Self built: Yes
Tested with official releases: Yes
(Persists with both self-built and official releases)
4GB FAT32 SD card
64GB FAT32 SD card

Tested official releases from 0.8.4 through 0.8.7 and it persists through all 3

Additional sysmodules: sys-clk and sys-ftpd, but the issue persists even with them removed

Homebrew installed: N/A

Krude, 2019-04-16:
Just tried it myself (AMS 0.8.7, FW 7.0.1) and can confirm, it hangs on the Nintendo Switch logo.

SciresM, 2019-04-19:
Resolved.

commit e77305f0 (2019-04-19): pm: on 7.0.0+, npns is launched in maintenance boot (closes #511)
commit 908de31a (2019-04-20): pm: on 7.0.0+, npns is launched in maintenance boot (closes #511)
closed 2019-04-20 by SciresM