Atmosphere-NX/Atmosphere #512: Ability to restart to RCM or shutdown from fusee-primary (when fusee-secondary isn't present)

aveao, 2019-04-17:
## Feature Request

### What feature are you suggesting?
#### Overview:
fusee-primary currently only allows you to try again to find fusee-secondary from microsd by pressing power.

My request is for an addition of options to reboot to RCM or shutdown, with the volume buttons.

#### Smaller Details: 
Unknown

Done already by hekate and various parts of AMS inside horizon, so implementation details should be trivial to determine to those who have a better understanding of those codebases.

#### Nature of Request:  
- Addition
  - Addition of certain original features or features from other community projects.

Hekate already has this, and is licensed GPLv2.

### What component do you feel this would best fit within?
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's custom bootloader.</br>

### Why would this feature be useful?
This might be useful to people (like me) who accidentally pushed the wrong payload and had to hold down power button to shut down (which leads to the screen flashing due to it trying to find secondary again, fwiw).

SciresM, 2019-04-17:
@aveao Sure, good idea. On another note, we should re-do how we reboot to RCM.

Right now, we just set the "use RCM mode" flag then do a PMC reboot. It would be friendlier to use the warmboot arb-writes to pwn the bootrom and add some patches, so that fusee will work if RCM is rebooted to on an ipatched unit. I kind of worry that this might cause a lot of confusion about whether fusee works (and I worry about people installing AutoRCM and bricking because they think it'll work), but I bet it'd be really nice for owners of patched units....

sand999, 2019-06-04:
Why not have a special color to show that their console is patched. That way they know not to put autorcm

DraconicNEO, 2022-03-22:
> @aveao Sure, good idea. On another note, we should re-do how we reboot to RCM.
>
> Right now, we just set the "use RCM mode" flag then do a PMC reboot. It would be friendlier to use the warmboot [...]

I think that seems like a cool idea although maybe it would also be a good idea to have an option to reboot to `SD:/payload.bin` for those people who are using modchips since at least for them it would be more convenient to Boot to Hekate than booting into RCM.

SciresM, 2022-03-22:
> I think that seems like a cool idea although maybe it would also be a good idea to have an option to reboot to `SD:/payload.bin` for those people who are using modchips since at least for them it [...]

1) please do not comment on issues from 2019.

2) it is not possible to reboot to payload on devices with modchips. That requires an exploit and we don't have one.

SciresM, 2025-04-03:
I don't know what the status of this issue was, but I'm closing it because fusee-primary no longer exists

closed 2025-04-03 by SciresM