Atmosphere-NX/Atmosphere #515: OFW 8.0 Homebrew running, but game wont start

lajax01, 2019-04-21:
OFW 8.0, with latest Atmosphere.
Boot up just fine.
 Homebrew running. I can install games, but in main no games start up.
Give me a various errors like, corrupted data, download again, etc.. 
What is missing?
(Still using lockpick generated keys from 7.0.1)

ZachyCatGames, 2019-04-21:
All of my legitimately purchased physical and digital games work without any issues 🤔

lajax01, 2019-04-21:
Point taken, but fck yourself. ;)
My legit titles wont start under latest atmosphere, only if I start a console in factory state.
In sort: no legit or pirate content wont start, only hb apps under album.

fincs, 2019-04-21:
> Homebrew running. I can install games, but in main no games start up.

> My legit titles wont start under latest atmosphere, only if I start a console in factory state.

???

lajax01, 2019-04-21:
Seriously? I cant say more about this issue.
Only hb apps works, nothing else.

DankanTsar, 2019-04-21:
Your pirate patches are a bit faulty. I think this is not a problem of atmosphere.

lajax01, 2019-04-21:
Any suggestions? 
I want play a particular game.. :/
Kip and sigpaches are fully working on Atmosphere on fw 8.0?

SciresM, 2019-04-21:
we don't support warez.

closed 2019-04-21 by SciresM