Atmosphere-NX/Atmosphere #516: Atmopshere

Austin70000, 2019-04-21:
## Bug Report

Atmosphere crashes at Nintendo Switch logo upon injecting Fusee Primary. Had just updated atmosphere on Switch SD for FW 8.0.0

Updated/changed CFW and payload
Can be reproduced upon every startup

[image]

### Crash Report

Crash screen attached.
### System Firmware Version

8.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2019-04-21:
Do you have maybe custom theme?

sorenbug, 2019-04-21:
This is most likely due to a custom theme. Delete any `01000000000010xx` folders under `/atmosphere/titles/`, then open NXThemes Installer and uninstall + reinstall your themes. Make sure to say "yes" when it asks you about re-dumping the home menu files.

Austin70000, 2019-04-21:
Thank you! Closing issue.

closed 2019-04-21 by Austin70000