Atmosphere-NX/Atmosphere #518: Fatal auto reboot after countdown

fennectech, 2019-04-22:
## Feature Request

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ If you do not have the information needed for a section, replace its contents with "Unknown". ]</br>
[ Lines between [ ] (square brackets) are to be removed before posting. ]

[ Please search for existing [feature requests](https://github.com/Atmosphere-NX/Atmosphere/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3A%22features%2Ffeature-request%22) before you make your own request. ]</br>
[ Duplicate requests will be marked as such and you will be referred to the original request. ]

### What feature are you suggesting?
#### Overview:
- [ Include the basic, high-level concepts for this feature here. ]</br>
Simple auto reboot timer in fatal.

#### Smaller Details: 
- [ allow fatal to automaticly reboot . ]</br>

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Change

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
[ If this is a feature for an end-user, how does it benefit the end-user? ]</br>
It would allow users of docked switches not to have to get up to reboot a fataled switch.

HookedBehemoth, 2019-04-22:
[ Lines between [ ] (square brackets) are to be removed before posting. ]

fennectech, 2019-04-22:
That better?

HookedBehemoth, 2019-04-22:
Yes

SciresM, 2019-04-22:
I don't see why this would be desirable. Hit a button...?

fennectech, 2019-04-22:
if the console is docked and im sitting back on the couch i'd rather wait a timeout than get up to hit the power button

SciresM, 2019-04-22:
Implement yourself, make it not the default, and use the setting "atmosphere!fatal_auto_reboot_interval", expressed in milliseconds.

I will accept your PR.

closed 2019-04-22 by SciresM

fennectech, 2019-04-22:
This aint bash      ill see if i can make it but i doubt i will be able to.