Atmosphere-NX/Atmosphere #525: How do i install Senran Kagura Feet fetish dlc?

Reiisgei77, 2019-04-25:
hi first of all dont judge me and also im using the latest version of atmosphere and i own a japanese cart of Senran Kagura but i dont know how could i get this dlc

ZachyCatGames, 2019-04-25:
> hi first of all dont judge me and also im using the latest version of atmosphere and i own a japanese cart of Senran Kagura but i dont know how could i get this dlc

open the eshop —> buy the dlc —> download the dlc 

also this isn’t an atmosphere issue :P

Reiisgei77, 2019-04-25:
yea but im using an european account

ZachyCatGames, 2019-04-25:
> yea but im using an european account

You can add a japanese account to your system and use that :P 

still not an atmosphere issue though

Reiisgei77, 2019-04-25:
**Hi to my friends at reswitched and hi to my bruddas at reiswitched**

fincs, 2019-04-25:
Closed due to off-topic.

closed 2019-04-25 by fincs

Reiisgei77, 2019-04-25:
:feelsbad:

jamescallen246, 2019-04-25:
You will need a Japanese account