Atmosphere-NX/Atmosphere #529: Atmosphere crashes when closing a title.

HaveSumFaith, 2019-04-25:
My situation:
New titles require the latest system to play, so i had to upgrade using choijur from 6.0 to 7.0 ( previously i had my system downgraded to 6.0 because my little brother updated it to 7.1 but it was working fine for months) So i have upgraded to 7.0 yesterday, and i have this weird issue: atmosphere crashes every time i close a game. No matter which game.  I have tried all 3 of the "pre configured bundles" at sdsetup. (kosmos defaults, reinx defaults and "minimal") Reinx is unusable, its incredibly buggy and unstable, kosmos with atmosphere is crashing, and the minimal package is holding out a little better, but it starts to act up after some time too.

Not sure what could be the problem, with the pre-configured package i got the sig patches as well...

Also, i have tried different payloads too, but it don't make a difference.
[image]

TuxSH, 2019-04-25:
Did you try latest Atomsphère release? (properly setting it up, *without* sdfiles)? https://github.com/Atmosphere-NX/Atmosphere/releases

TuxSH, 2019-04-25:
> Reinx
>  sig patches as well...

We don't support warez.

closed 2019-04-25 by TuxSH

HaveSumFaith, 2019-04-25:
@TuxSH OMG once of the programming wizards replied to my post :O But Yes sir, i did. Deleted everything except my nintendo folder. See, i have over 100gb of games on my nintendo folder, if i delete it im gonna have to spend a whole day reinstalling and moving them back :(

HaveSumFaith, 2019-04-25:
Mr.Wizard, please don't close it :( I need help.

fincs, 2019-04-25:
Piracy or warez is *not* supported by Atmosphère. "Your" over 100gb of games, plus the fact that you have used warez oriented distributions of Atmosphère (including signature check patches) is a giveaway of the fact that they are all pirated.

TuxSH, 2019-04-25:
If you have bought the games you can redownload them (assuming you're not on a metered connection).

In fact you do not need to touch the Nintendo folder at all.

fennectech, 2019-04-25:
Redownloading on a /fast/ connection over a hundred gigs of games takes
24h+.  All of that time most games cannot be played as that will stop
downloads.
-- 
FennecTECH

TuxSH, 2019-04-25:
With the ethernet stuff capped at 50MB/s (which is a speed you can easily get with proper optical fiber) this just takes a good night.

Also don't tell me you need to play all 100 in the same day :).

HaveSumFaith, 2019-04-25:
Okay. I didn't know you guys don't support the ol " take what ya can and don't give it back" philosophy, but i'm genuinely need some help. As i said, the "minimal" package from sdsetup works best, and it doesnt freeze when i close a game (during a fresh install) but after a couple hours, it freezes just like the other packages.

DavidBuchanan314, 2019-04-25:
Did you even try the first troubleshooting step that the "programming wizard" told you to try?

If you haven't replicated the issue with a vanilla, up-to-date atmosphere install, I'm not sure what you're doing here.

HaveSumFaith, 2019-04-25:
@DavidBuchanan314 Not sure what you mean? He suggested me to try the latest build, which as a human with a brain, was my first thought. Of course i tried it man, it was the first package.

fincs, 2019-04-25:
Locking due to the warning that piracy is not supported being repeatedly ignored.