Atmosphere-NX/Atmosphere #531: Massive installed titles corrpution

ParodZetec, 2019-04-28:
When I updated to 8.0.1 and got AMS 0.8.8 some titles stopped working showing an "The software was closed" error, which appeared inmediately when the title just started to launch. This happens randomly with any type of titles (there were even legitimally bought games and some of them were installed on NAND, so it couldn't be SD corruption -or could be-). I have all 8.0.0 sigpatches installed and work correctly (as they allow me to launch titles). When the error happens, the title icon is well but in the left side of the game title there is an square with an exclamation symbol, and refuses to launch saying there are damaged data. I'm using Windows so it can't be the MacOS archive bit issue (I tried the fix provided in Hekate anyway but nothing worked).

No modifications were done to atmosphere code in use, I always use the default config to avoid messes (I updated to AMS 0.8.9 and same)

As this happens randomly I certainly can't know how is the problem reproduced 

No crash dump was done, as Atmosphere even seems to not have registered it

This happened on 8.0.1 firm. Launched Atmosphere via official TegraRCMgui and RCMLoader dongle.

TuxSH, 2019-04-28:
We don't support warez.

closed 2019-04-28 by TuxSH

ParodZetec, 2019-04-28:
Okay. I'm sorry for that. I just posted this because it happened with bought titles too.