Atmosphere-NX/Atmosphere #533: A way to patch the value of the analog sticks dead zone

ShinKaz, 2019-05-04:
would it be possible to patch the value of the analog sticks dead zone upon booting ?
like in system_settings.ini or something.

I know a lot of ppl are having issues with joycons or pro pad analog sticks dragging in a given direction and recalibration in the setting menu doesn't change anything in this issue :(

thanks a lot in advance :)

fincs, 2019-05-04:
That's not really a problem that can be solved in software; it's a hardware issue, and if I understand it correctly you're supposed to blow some compressed air under the joystick caps.

closed 2019-05-04 by fincs

DavidBuchanan314, 2019-05-04:
@ShinKaz this feature could possibly be implemented in hid-mitm:

https://github.com/jakibaki/hid-mitm

fincs, 2019-05-04:
But for what reason? The real fix to this issue is to fix the hardware.

DavidBuchanan314, 2019-05-04:
My understanding is that the potentiometers inside the joysticks get worn down, it's not just a question of removing dust. Adjusting the deadzone seems like a reasonable workaround for those not yet in a position to replace the joystick.

Xamanthas, 2019-05-04:
Agreed, there are some pics out there showing the damage that occurs to the sticks. Only replacing them can fix it normally

ShinKaz, 2019-05-04:
why ? because increasing the deadzone by like 10% would not drastically change the user experience but fix his issues.

it's an option on PC, emulators but console makers always fail to add, hence the request.

tbh, it's a widespread issue on switch and opening up you controllers all the time isn't the best thing either. I didn't play my switch a lot since I got it, my 360 controller showed the same issue after using it maybe a 100x that, this is kinda ridiculous.

ShinKaz, 2019-05-04:
> @ShinKaz this feature could possibly be implemented in hid-mitm:
> 
> https://github.com/jakibaki/hid-mitm

yes but this adds lag that could not be there when uting OG controllers.

DavidBuchanan314, 2019-05-04:
If you're a competitive gamer who cares about 1 extra frame of input latency, you should probably buy a new controller :P

friedkeenan, 2019-05-06:
Just want to say that the latest commits of hid-mitm have a deadzone