Atmosphere-NX/Atmosphere #538: Cuphead crashes when starting new game

nbalonso, 2019-05-09:
## Bug Report

### What's the issue you encountered?

New Cuphead game with no previous saved file. Selected NEW file and character. Expected to see the intro video but the game crashes with 

> The software was closed because an error occurred.

Tried setting `dmnt_cheats_enabled_by_default` to `0` but still consistently crashes.

Might be related to video/audio decompression+decoding since I never get to hear the intro music. Music and animations in the main menu are fine

### How can the issue be reproduced?

Is like the first 20 seconds of https://youtu.be/nKzXOwswTLs, then

> The software was closed because an error occurred.

### Crash Report

https://gist.github.com/nbalonso/6772666562189d5431c764d2759123a4

### System Firmware Version

8.0.1

### Environment?

- What bootloader: fusèe-primary
- Atmosphère Official release version 0.8.9
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: EdiZon

### Additional context?

- Additional info about your environment: if needed, a dump _thread_info.bin file is also generated when the game crashes

SciresM, 2019-05-23:
Can't reproduce. Successfully created a new save file in cuphead on 8.0.1 + AMS 0.8.10.

Will re-open if this becomes reproducable.

closed 2019-05-23 by SciresM

nbalonso, 2019-05-29:
turns out data was corrupted. Found out via system settings, data management, software

Redownloading the game fixed it

sorry for the wrong report

SciresM, 2019-05-30:
No problem.