Atmosphere-NX/Atmosphere #540: Cant build AMS on Ubuntu

2g4y1, 2019-05-10:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Failed to build on Ubuntu 18.04 LTS, latest commits

### How can the issue be reproduced?

Check AMS out recursive with submodules and build.
Tested on 2 machines, same result

### Crash Report

-

### System Firmware Version

- latest git commits

### Environment?

Ubuntu 18.04.2 LTS (GNU/Linux 4.15.0-20-generic x86_64)

### Additional context?

Build log.
[putty.log](https://github.com/Atmosphere-NX/Atmosphere/files/3167443/putty.log)

SciresM, 2019-05-10:
Update libnx, you need latest libnx master to build ams-master.

closed 2019-05-10 by SciresM

austin5623, 2019-05-10:
[ L i n e s b e t w e e n [ ] (s q u a r e b r a c k e t s) s h o u l d b e r e m o v e d b e f o r e p o s t i n g. ]