Atmosphere-NX/Atmosphere #543: Game, Croixleur Sigma, does not work on atmosphere

Elionna, 2019-05-12:
## Bug Report

### What's the issue you encountered?

Croixleur Sigma does not run on atmosphere. When launching, the game crashes during the loading logo and returns to the switch homescreen. No other game exhibits this problem and when running in pure stock the game works fine.</br>
No changes were made to atmosphere.</br>

### How can the issue be reproduced?

Purchase Croixleur Sigma from the eshop and try to play it, the game will crash upon launching </br>

### Crash Report

https://gist.github.com/Elionna/4b0ac41db098827eb27a738e99b7fade

### System Firmware Version

8.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release: Yes
  - 0.8.9
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: NXShell

### Additional context?

- Additional info about your environment:
- N/A

SciresM, 2019-05-23:
This is a fs.mitm issue. After a lot of debugging, I've determined that the crash is a NULL deref because a global pointer that is supposed to be set in another thread (related to file code) is not being written correctly. Not sure why this is happening, working on it.

I have confirmed that atmosphere with fs.mitm disabled boots the game, though, so I'm honing in on a resolution.

SciresM, 2019-05-28:
@Elionna This game is written poorly.

There is a race condition that arises because the FS reader thread does not synchronize with the main thread (no locks are used before accessing global variables that the FS reader writes to).

The overhead from running fs.mitm on top of FS causes this race condition to hit most of the time on the very first load, which causes a null deref.

It's a game code issue, not an atmosphere issue...I think the issue could also occur on non-stock if the SD card was a little slower than expected...

Anyway, here's an IPS patch that fixes the problem (it simply makes the main thread verify the FS reader state is non-NULL). Save it at `/atmosphere/croixleur_sigma_race_fix/C8568FF6F80F2BF0FEA8048D7AEF9864.ips`.

```
49 50 53 33 32 00 00 14 40 00 04 F0 BE 08 14 00 23 10 00 00 88 FD 7B B0 A9 E0 07 01 A9 E2 0F 02 A9 E4 17 03 A9 E6 1F 04 A9 E8 27 05 A9 EA 2F 06 A9 EC 37 07 A9 EE 3F 08 A9 F0 47 09 A9 F2 4F 0A A9 00 30 40 F9 1F 00 00 F1 C1 00 00 54 40 00 80 D2 00 BC 70 D3 61 01 00 D4 E0 0B 40 F9 F9 FF FF 17 01 00 40 F9 3F 00 00 F1 20 FF FF 54 FF 07 41 A9 E2 0F 42 A9 E4 17 43 A9 E6 1F 44 A9 E8 27 45 A9 EA 2F 46 A9 EC 37 47 A9 EE 3F 48 A9 F0 47 49 A9 F2 4F 4A A9 FD 7B D0 A8 C0 03 5F D6 45 45 4F 46
```

closed 2019-05-28 by SciresM

Elionna, 2019-06-01:
Thank you very much for looking into and fixing this!!!

SciresM, 2019-06-01:
@Elionna of course -- thank you for providing a thorough report with crash report attached. It was very helpful for diagnosing the problem :)

theedward1, 2019-06-04:
@SciresM I am trying to use the included IPS patch but it still seems to crash (on latest atmosphere), is there any special way I should be saving that file or settings I shoud be using? 

This is what I am using: 
[croixleur_sigma_race_fix.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3251831/croixleur_sigma_race_fix.zip)

ginkuji, 2019-06-24:
having the same problem. I can't get the ips to work. :(
System Firmware: 8.1
Bootloader: fusee
Official release: Yes
0.9.1
Do you have additional kips or sysmodules you're loading: N/A

SciresM, 2019-06-24:
@ginkuji Is your copy of Croixleur updated? I didn't bother making an IPS for the base game when I did the above, only for the latest update.

ginkuji, 2019-06-24:
@SciresM  Im using the latest patch 4 Croixleur Sigma[01000F0007D92800][US][v262144]

SciresM, 2019-06-24:
Oh, meh, game was updated since the IPS was made.

Wait for someone to make an updated IPS, or downgrade to the version the IPS patch targets.

I'll make an updated patch eventually if nobody else does, but it's not very low priority at the moment.

ginkuji, 2019-06-24:
Thank you! :) looking forward to that in the future.

ginkuji, 2019-06-24:
I tried it with patch 2-4 non of them works. :(

HookedBehemoth, 2019-11-25:
I've tried updating the IPS to the current version (1.0.5)
The printf you override is at 0x14ac
.rodata starts at 0x242000
build id: 9A289E6D9878961DA6ACB22CC45EF7E800000000000000000000000000000000

Sadly the game still doesn't start on atmosphere.

[9A289E6D9878961DA6ACB22CC45EF7E8.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3888958/9A289E6D9878961DA6ACB22CC45EF7E8.zip)

HookedBehemoth, 2019-12-08:
It works again on Atmosphère 0.10.0.

Elionna, 2019-12-11:
> It works again on Atmosphère 0.10.0.

The latest update is 1.0.6, when you have the time can you please update the IPS patch for that version?

SciresM, 2019-12-11:
I believe the game actually works without patches on Atmosphere 0.10.x. Can you confirm/deny?

Elionna, 2019-12-11:
I'm running an older version of atmosphere, will update and let you know by this Sunday.

Elionna, 2019-12-16:
@SciresM works perfectly on atmosphere 0.10.1 without patches, don't know what you did, but thanks for fixing this pesky game!

MmmmmDonuts, 2024-07-26:
Issue present in Atmosphere 1.7.1
Latest game version 1.0.8, update v524288 (last update was 2022-05-26, game can be considered final version)
Title ID: 01000F0007D92000
Build ID: 24FFB81F6298365E75B84005BF088BDE00000000000000000000000000000000