Atmosphere-NX/Atmosphere #544: Update 8.9 Won´t start. Error 2444-0002

Klauserus, 2019-05-15:
## Bug Report

### What's the issue you encountered?

Send Payload Hekate from PC to Switch. Lanch Button and CFW. I see the Sept Logo, Initializing, I see Nintendo Logo, is see the Switch Logo short times and then Atmo Error (see below).

The funny thing. When I press the volume button, I see welcome (font in the right pane), then an atmosphere logo and it boots normally with Atmosphere into the OS (7.0.1 (Atmo 8.9) on seetings.

I dont know what i can do next. I delete my SD Card. Format them new. copy fresh Atmosphere Data into it. I delete all Data, copy new and i overwrite them with the new one. Nothing works.

### How can the issue be reproduced?

### Crash Report
Atmosphère Fatal Report (v1.0):
Result:                          0x5BC (2444-0002)

Title ID:                        0100000000000037
Firmware:                        7.0.1 (Atmosphère 0.8.9-master-8c3dae84)
General Purpose Registers:
    PC:                          00000000
Start Address:                   00000000
Stack Trace:

### System Firmware Version

7.0.1

### Environment?

- Hekate 4.10.1 and lanch cfw
- SD Card 256GB SanDisk U3 with FAT32
- es_patches on SD
- kip_patches yes (folder default_nogc and fs_patches)
- various titles. are these relevant?

Thank you for that great peace of Software. It makes the Switch usefull :) and thx for help

SciresM, 2019-05-15:
This is a version mismatch error -- you are not actually updating atmosphere fully. Please ensure you are correctly updating atmosphere (fully extracting release zip).

closed 2019-05-15 by SciresM

Klauserus, 2019-05-16:
Hi. I know that is a mismatch, but i have extract the actually zip file from here and wrote this to the issue post: I dont know what i can do next. I delete my SD Card. Format them new. copy fresh Atmosphere Data into it. I delete all Data, copy new and i overwrite them with the new one. Nothing works.

All the delete, overwriting sh.., never will help me.

I thought that when I describe when the boot process stops, you can tell me which file is affected.

SciresM, 2019-05-16:
Boot via fusee-primary, and not whatever other bootloader you're using. Whatever the other thing you're using is is using a stale fusee-secondary.