Atmosphere-NX/Atmosphere #566: Compilining make dist error compiling (boot_calibration.o)

vaccarieli, 2019-06-15:
I have installed all dependencies to compile AMS as i just installed DevKitPro succesfully along with Pacman, Unzip and Libx, I used the devkitPro Updater v3.0.3 this is my log output:

aarch64-none-elf-g++ -MMD -MP -MF /home/elios/atmosphere/stratosphere/boot/build/boot_calibration.d -g -Wall -O2 -ffunction-sections -march=armv8-a -mtune=cortex-a57 -mtp=soft -fPIE -DDISABLE_IPC -DATMOSPHERE_GIT_BRANCH=\"master\" -DATMOSPHERE_GIT_REV=\"master-e5e605a\" -I/home/elios/atmosphere/stratosphere/boot/include -I/home/elios/atmosphere/stratosphere/boot/../../common/include -I/opt/devkitpro/portlibs/switch/include -I/opt/devkitpro/libnx/include -I/home/elios/atmosphere/stratosphere/boot/../libstratosphere/include -I/home/elios/atmosphere/stratosphere/boot/build -D__SWITCH__ -fno-rtti -fno-exceptions -std=gnu++17 -c /home/elios/atmosphere/stratosphere/boot/source/boot_calibration.cpp -o boot_calibration.o
C:/Users/elios/atmosphere/stratosphere/boot/source/boot_calibration.cpp: In function 'Result GetBatteryVendorImpl(u32*)':
C:/Users/elios/atmosphere/stratosphere/boot/source/boot_calibration.cpp:58:38: error: 'FsBisStorageId_CalibrationBinary' was not declared in this scope
     Result rc = fsOpenBisStorage(&s, FsBisStorageId_CalibrationBinary);
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/elios/atmosphere/stratosphere/boot/source/boot_calibration.cpp: In function 'Result GetBatteryVersionImpl(u32*)':
C:/Users/elios/atmosphere/stratosphere/boot/source/boot_calibration.cpp:79:38: error: 'FsBisStorageId_CalibrationBinary' was not declared in this scope
     Result rc = fsOpenBisStorage(&s, FsBisStorageId_CalibrationBinary);
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [/opt/devkitpro/devkitA64/base_rules:14: boot_calibration.o] Error 1
make[2]: *** [Makefile:135: build] Error 2
make[2]: Leaving directory '/home/elios/atmosphere/stratosphere/boot'
make[1]: *** [Makefile:10: boot] Error 2
make[1]: Leaving directory '/home/elios/atmosphere/stratosphere'
make: *** [Makefile:21: stratosphere] Error 2

shchmue, 2019-06-15:
you'll need to build libnx master and use that.

vaccarieli, 2019-06-15:
> you'll need to build libnx master and use that.

Thanks, it worked now.

closed 2019-06-15 by SciresM