Atmosphere-NX/Atmosphere #573: sysNAND boot stalls after emuMMC shutdown

aedalzotto, 2019-06-17:
## Bug Report
### What's the issue you encountered?
After using emuMMC and shutting down the Switch stalls its normal boot and needs to be shut down in the "Nintendo Switch" splash screen. Then, retrying a normal boot works again.

### How can the issue be reproduced?
Boot emuMMC (separate partition mode) with fusee-primary.
Shutdown (hold power button -> power options -> turn off)
Power up without payload
Now, the Switch will hang forever in the Nintendo Switch splash screen. Short press power to shut it off.
Press power again to normal boot (no payload).
The Switch will boot properly.

### System Firmware Version

8.0.1

hippydave, 2019-06-18:
I had this too.

m4xw, 2019-06-18:
This doesn't happen if you inject with a payload?

m4xw, 2019-06-18:
Regulator issue, related to https://github.com/m4xw/emuMMC/issues/2
Not a atmo issue.

closed 2019-06-20 by SciresM