Atmosphere-NX/Atmosphere #575: Atmosphere 0.9.0 fail to boot on 3.0.0

Akorax, 2019-06-18:
## Bug Report

### What's the issue you encountered?
Disclaimer: This is the first time I try to run any exploits/CFW on the Switch.
1. I followed the [nh guide](https://nh-server.github.io/switch-guide/user_guide/getting_started/)
2. Atmosphere ran fine a couple of times. While checking things out I tried starting homebrew menu via the gallery.
3. I rebooted and enabled AutoRCM in Hekate.
4. Now when trying to boot with Atmosphere it just fails and return to Hekate through autoRCM.

No changes were made to Atmosphere and reverting back to 0.8.10 fixes this issue.

### How can the issue be reproduced?
For me I just need to install version 0.9.0 and it fail to boot from Hekate.

### Crash Report

No error screen or crash report were generated

### System Firmware Version

3.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate - CTCaer mod
- Official release or unofficial build: Official
  - 4.10.1
- Do you have additional kips or sysmodules you're loading: Nothing else than stated in the guide.
- Homebrew software installed: I suppose ftpd, Checkpoint ,NX-Shell, NxThemesInstaller counts for this.

### Additional context?

- Additional info about your environment:
- Repeat: Reverting to 0.8.10 fixes the issue.
- I have tried completely deleting all Atmosphere related files on the SD card and downloading a new copy of 0.9.0 but the issue persists.

SciresM, 2019-06-18:
Does issue occur launching with fusee from official release instead of hekate?

SciresM, 2019-06-18:
Also: "fails to boot" can mean a lot of things.

What, specifically, happens?

Akorax, 2019-06-18:
> Also: "fails to boot" can mean a lot of things.
> 
> What, specifically, happens?

You mean use the provided reboot_payload.bin? I will try that.
What exactly happens is that after the black and white Atmosphere logo dissapear the console screen goes black (here I would expect the black and white Switch logo) and then restarts. Since I have autoRCM enabled that's where I end up.

No idea if this was just screenflicker but once I saw the Atmosphere logo go red before the screen went black.

SciresM, 2019-06-18:
No, I mean the official fusee binary, not reboot payload.

Akorax, 2019-06-18:
> No, I mean the official fusee binary, not reboot payload.

Works like a charm! Some issue with Hekate then? I take this opportunity to thank you for the help on the issue and also for bringing so much to the community and gamers worldwide.

CTCaer, 2019-06-18:
What's your config for hekate?

Akorax, 2019-06-18:
> What's your config for hekate?

I hope I'm correct in assuming you want to see the hekate.ipl.ini contents. If not please guide me to the information you need. I have not read enough yet to begin doing any sort of modifications other than instructed.

Here's Hekate.ipl.ini
[config]
autoboot=0
autoboot_list=0
bootwait=3
verification=1
backlight=100
autohosoff=0
autonogc=1
{------ Atmosphere ------}
{Pick this option to launch CFW.}
[Atmosphere FSS0]
fss0=atmosphere/fusee-secondary.bin
kip1=atmosphere/kips/*
{ }
{------ Stock ------}
{NOTE: This option does not launch CFW.}
[Stock]
fss0=atmosphere/fusee-secondary.bin
stock=1
{ }

CTCaer, 2019-06-18:
What atmosphere/kips/* has?
I assume not the atmo kips, because these override the ones in fss0.

Akorax, 2019-06-18:
> What atmosphere/kips/* has?
> I assume not the atmo kips, because these override the ones in fss0.

Everything is default pretty much since I just started with all of this. The "kips" folder doesn't exist, only "kip_patches" which has the "nogc"-folder in it and in there 14 files that are provided from the start with AMS 0.9.0.

CTCaer, 2019-06-18:
It's an easy guess then:

Exo receives garbage on emummc extra cfg, because v4.10.1 never sets anything there.
Exo also doesn't look for a changed bit on .flags, in order to enable the emummc configuration.
And thus it fatals.

So consider ams 0.9.0 incompatible with hekate v4.10.1 and lower.

closed 2019-06-18 by SciresM