Atmosphere-NX/Atmosphere #576: 'Incognito' application crashes (with emummc)

dbenoy, 2019-06-19:
### What's the issue you encountered?

A tool called 'incognito' has been recommended to me. I'm told it eliminates identifiable information from the Switch so it can't be banned for using CFW, but it produces errors and crashes. I suspect it may have to do with emummc? Although I leave my system memory untouched so I haven't confirmed that.

### How can the issue be reproduced?

Run the nro from here: https://github.com/blawar/incognito

The following errors show up right away:
"error: failed to open cal0 partition"
"error: failed reading calibration data"

The process seems to fail.

And if you press X to 'verify prodinfo NAND' you get a crash report.

### Crash Report

[01560921819_0100000000000023.log]

### System Firmware Version

1.0.0

Then I updated the emunand to 8.0.1 and got the same results.

### Environment?

fusee-primary.bin
Atmosphère 0.9.0-master-aa289410

SciresM, 2019-06-19:
Not supported.

dbenoy, 2019-06-19:
Oh that's too bad. Thanks anyway.

closed 2019-06-19 by dbenoy