Atmosphere-NX/Atmosphere #578: Switch dosent start

XiNaru, 2019-06-19:
Hello,
i had updated my switch to 8.1.0 first, then i updated today the athmosphere. but after injecting the new fusee, and an loading screen from athmosphere, blackscreen
nothings happens anymore.....

ELY3M, 2019-06-19:
I have same issue on my switch with 6.2 fw   try again with reinjecting payload.

Coelacanth13, 2019-06-19:
I have the same issue: running the newest update on 8.1.0 I get to the  atmosphere logo screen then a black screen after there.
Have tried injecting multiple times, same issue each time.

Ch0cola, 2019-06-20:
How you do this update? Through system settings or with ChoiDujourNX?

There are some reports of users that use Choi to update and now can't boot.

Specced, 2019-06-20:
My understanding is that updates can be applied with system update if you are 100% certain you are inside the EmuMMC.

ELY3M, 2019-06-20:
I updated my other switch to 8.1.0 offical way.  I was able to start ams on it fine.
Then I updated ams on my 6.2 switch and It kept going back to rcm mode after black and white logo screen.  Strange case here.  I will try new atmosphere with fix when I get home.

closed 2019-06-20 by SciresM