Atmosphere-NX/Atmosphere #579: Emunand Issue

mrdude2478, 2019-06-19:
I've just updated to Atmosphere 0.9.1

Firmware on Emunand 8.1.0 - emunand freezes on Nintendo Logo.
Reinstalled firmware 8.0.1 - emunand works fine.

Seems there's an issue using emunand which has firmware 8.1.0 installed.

8.1.0 fimware on emunand - works fine with SXOS.

d3st1nyh4x, 2019-06-19:
works fine on ams too

mrdude2478, 2019-06-19:
No, it works fine with 8.0.1 firmware on both SXOS and Atmosphere. On 8.1.0 firmware - it works only on SXOS, Atmosphere freezes (emunand works ok on 8.0.1 though).

d3st1nyh4x, 2019-06-19:
im using emunand on 8.1 right now

mrdude2478, 2019-06-19:
Well - it was working before - now it doesn't - I've gone back to 8.0.1 and it works again.

aedalzotto, 2019-06-19:
Update fusee-primary and sept too

mrdude2478, 2019-06-19:
Yep, I've done that. I can boot into Atmosphere 0.9.1 with 8.0.1 firmware on emunand  - just not 8.1.0.

Ch0cola, 2019-06-20:
How you do this update? Through system settings or with ChoiDujourNX?

There are some reports of users that use Choi to update and now can't boot.

SciresM, 2019-06-20:
Can't reproduce, booting successfully into 8.1.0 emunand. 

Are you using a build you built yourself? Closing for now.

closed 2019-06-20 by SciresM

Ch0cola, 2019-06-20:
I'm managed to reproduce this issue after some boots on emunand.

I updated to atmos 0.9.1 with emunand on 8.0.1, I boot sucesfully a lot of times but now the emunand stuck on Switch logo (after show the icon of atmos on black and white).

There are any way to get logs or some usefull info about what produce this issue?

SciresM, 2019-06-20:
@Ch0cola that sounds more like emummc issue than ams issue.

Ch0cola, 2019-06-20:
@SciresM It's possible, seems like a random data corruption after reboot

mrdude2478, 2019-06-20:
> How you do this update? Through system settings or with ChoiDujourNX?
> 
> There are some reports of users that use Choi to update and now can't boot.

Yep I updated through ChoiDujourNX, however that emunand still works with SXOS - so the nand is fine.

hippydave, 2019-06-20:
Any custom sysmodules? I read that some aren't working on 8.1.0

mrdude2478, 2019-06-20:
> Any custom sysmodules? I read that some aren't working on 8.1.0

You're right - I just deleted the Atmosphere/titles folder and contents - now it's booting. I'll have a look and add 1 at a time to see what's causing the issue.

mrdude2478, 2019-06-20:
OK, I found the problem: Delete folder

0100000000000352

If you have emuiibo installed, Atmosphere fails to boot.

Thanks.