Atmosphere-NX/Atmosphere #582: Black screen

zany130, 2019-06-20:
## Bug Report

### What's the issue you encountered?
booting with the fussgee payload shows a messages staying welcome to atomowhphere then a black screen</br>
No chages directly download from the release page</br>
No changes</br>

### How can the issue be reproduced?

insert and sd card with the atmosphere files and boot into rcm. Inject the fusegelee payload. You will get a welcome message and black screen </br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report
N/A</br>

### System Firmware Version

8.1.0</br>
### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: officiall release 
- Do you have additional kips or sysmodules you're loading:no
- Homebrew software installed:  none

### Additional context?

- Additional info about your environment:
- I have also tried the kosmos pack and same issue tried a new sd card same issue

SciresM, 2019-06-20:
It sounds like you didn't extract fusee-secondary correctly to the SD.

zany130, 2019-06-21:
You're right  that fixed it thanks  I was even able to get kosmos working by replacing  that as apparently they still haven't updated it although it crashes but again that's probably because it needs to be updated thanks though

closed 2019-06-21 by zany130

chubbysocks, 2019-09-19:
Can you please tell me step by step how you done it? I'm having a similar problem.

chubbysocks, 2019-09-19:
> It sounds like you didn't extract fusee-secondary correctly to the SD.

Hi can you explain what you mean by this? I'm having the same problem, many thanks.

deadair3210, 2019-09-19:
0. Take backup of SD card
1. Download latest version
2. Extract to the SD card, overwriting everything

chubbysocks, 2019-09-19:
> 1. Take backup of SD card
> 2. Download latest version
> 3. Extract to the SD card, overwriting everything

Before when I attached the USB C to my PC the switch would recognise it.... now it won't.