Atmosphere-NX/Atmosphere #583: Backup BIS keys on ipatched consoles

shchmue, 2019-06-20:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Save BIS keys to SD on boot if RCM ipatch detected</br>

#### Smaller Details: 
- fusee-secondary can read the appropriate fuses and already has the unused function `derive_bis_key`</br>

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's custom bootloader.</br>

### Why would this feature be useful?
This will increase the recoverability of softbricked patched consoles - NAND could ultimately be rebuilt with correct encryption and written by whatever hardware can mount a Switch eMMC (another Switch, Raspi, etc) and fits in with the philosophy that led to auto PRODINFO backups</br>

ckurtz22, 2019-06-27:
bump. This is a lot more relevant now that ipatched units are finally able to launch cfw. Even more so since it has been discovered that supernag is impossible to remove from these consoles without biskeys.

SciresM, 2019-07-01:
Yes, this is good to do and will be included in 0.9.3.

noirscape, 2019-07-10:
I would also suggest expanding this to dump the Device Key and boot0/boot1.

Device Key since BIS keys are derived from that and `boot0`/`boot1` because we can't fully recreate them (granted, keyblobs aren't used right now, but they could be in the future).

ckurtz22, 2019-07-10:
How does ChoiDujourNX get BOOT0/1 if we can't recreate them?

shchmue, 2019-07-10:
we're getting in the weeds here, but it only writes the relevant parts of them, the non-console-unique parts that are dictated by firmware version, and leaves the console-unique parts alone

SciresM, 2019-07-10:
Meh, "device key" -- requires dumping from elevated context versus bis keys.

Also, any console unique part of boot0 that has ever been used can be recreated.

commit 362ee3cd (2019-08-06): autobackup: dump bis keys (closes #583).
closed 2019-08-06 by SciresM