Atmosphere-NX/Atmosphere #588: Stuck at "sept" logo

JhonJhonny, 2019-06-21:
## Bug Report

### What's the issue you encountered?

I can't go beyond the sept splash logo
havent'made any changes

### How can the issue be reproduced?

Clean MicroSD formatted in Fat32, transferred the content of the atmosphere 0.9.1 in the sd, injected the payload, the welcome writing comes up, then the sept splash logo appear and can go beyond (its not a black screen, i'm stuck at the sept logo)

### Crash Report

No crash log present 

### System Firmware Version

8.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: Official release version 0.9.1
 
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Nothing

### Additional context?

- Additional info about your environment:

Skonikol, 2019-06-21:
If you solsuite macos, then try to fix archive bits via hekate.

JhonJhonny, 2019-06-21:
i'm not using macos, should i do that anyway? and how to?

Skonikol, 2019-06-21:
Try it, it won't hurt. It helped me.

JhonJhonny, 2019-06-21:
Tried, but same thing, stuck at the sept logo, thanks anyway

SciresM, 2019-06-21:
Make sure you have fusee-secondary at sept/payload.bin

closed 2019-06-21 by SciresM

JhonJhonny, 2019-06-21:
Obviously i have that as i downloaded the repository and put in the sd card everything.. i dont understand why you had to close the issue as i haven't resolved anything, any other advice?

SciresM, 2019-06-21:
"Downloaded the repository" -- what? sept logo stuck is user error due to bad layout of files on SD card.

JhonJhonny, 2019-06-22:
I downloaded the atmosphere 0.9.1 from github and extracted directly in my sd card formatted in fat32, i cant see how it could be a bad file layout

Just for the purpose of trying i tried the atmosphere 0.9.0, same thing done, extracted directly in the fat 32 sd, injected the fusee primary, welcome message and instead of showing the sept logo it show a black screen

meeces2911, 2019-07-05:
@JhonJhonny if you still haven't resolved this, then its likely either your sd card is corrupt/bad/broken or you are not safely removing/dismounting the sd card before removing it from the pc.
(I had a similar problem when I tried ftp'ing it to the sdcard the first time around)

tycho01, 2019-07-23:
> Make sure you have fusee-secondary at sept/payload.bin

Thanks, this led me in the right direction. It turned out my SD card was full and there was a 0-byte file where this payload should have been.