Atmosphere-NX/Atmosphere #589: Modules that disable cal0 protection make the emuNAND to not boot

EmptyAirSwap, 2019-06-21:
## Bug Report

Modules that disable cal0 protection make the emuNAND to not boot

### What's the issue you encountered?

Modules that disable cal0 protection (to install incognito mainly) make the emuNAND to not boot

### How can the issue be reproduced?

Copy the modules that disable cal0 to the sd and with emunand enabled, inject fusee primary

### Crash Report

This issue doesn’t crash 

### System Version Information 

7.0.1 (I think all firmware) 

### Enviroment
 
Atmosphère 0.9.1 

### Additional context?

Nothing

WerWolv, 2019-06-21:
Sounds like you just tried replacing ams_mitm with rnx_mitm which obviously fails. Partway because I think emuMMC depends on stuff there? Not quite sure about that

SciresM, 2019-06-21:
"Using unsupported modules causes stuff to not work"

Okay.

SciresM, 2019-06-21:
Rather than whatever vague modules you're using, just build latest ams mitm without the prot check if you must.

closed 2019-06-21 by SciresM

EmptyAirSwap, 2019-06-21:
Thanks @SciresM and @WerWolv